MichaelGrupp/evo #159: euroc ground truth file format

cxt98, 2019-04-18:
**Description:**

I just run a test on Euroc dataset and find evo_xxx commands fail on recognizing ground truth files. I have checked previous issues about euroc testing but their problems are not related.

**Command:**
```
evo_traj euroc ../LearnVIORB/KeyFrameNavStateTrajectory.txt --ref=../dataset/euroc/V1_02_medium/mav0/state_groundtruth_estimate0/data.csv -p
```

**Console output:**
```

[ERROR] EuRoC MAV state ground truth must have 17 entries per row and no trailing delimiter at the end of the rows (comma)

```

**Additional files:**
Please attach all the files needed to reproduce the error. I attached a zip file
[data.csv.zip](https://github.com/MichaelGrupp/evo/files/3092805/data.csv.zip)

The ground truth file I tested on is just the data from euroc V1_02_medium scene.

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.4.1
* Python version shown by `evo pkg --pyversion`: 3.4.3
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): ubuntu 14.04
* did you change the source code? (yes / no): no, just installed by pip
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s",
    "global_logfile_enabled": false,
    "plot_backend": "Qt5Agg",
    "plot_export_format": "pdf",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_multi_cmap": "none",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true
}

```

MichaelGrupp, 2019-04-18:
The "data.csv" file you attached is okay, I ran `evo_traj euroc data.csv` and it worked.

Which format is `KeyFrameNavStateTrajectory.txt`? I think that's the problem. `evo_traj euroc` expects all files to be in `euroc` format.

If `KeyFrameNavStateTrajectory.txt` is in the TUM format, you could convert the ground truth file to that format using `--save_as_tum`. Then you can just use `evo_*** tum` without worrying about the format.

cxt98, 2019-04-18:
Yeah it's my fault. My KeyFrameNavStateTrajectory.txt is likely the euroc format with 17 columns, but not a csv file with commas as delimiters, and the ground truth file is correct. It works after I convert the text to csv file and also convert both to tum file. Maybe you can perfect the error message because it always tells the ground truth file format is erroneous, rather than other files.

closed 2019-04-18 by cxt98