MichaelGrupp/evo #160: full trajectory error data and scale function

cumtchenchang, 2019-05-01:
Thank you for your good work. If your work could add this two functuions, sincerely.
1) Could the full trakectory error be saved with the index or timestamp, such as as txt format. the saved data could be used  for deep analysis.
[image]

2) Could scale comparsion be added for stereo SLAM or VIO, as shown in the figure below.

[image]
Raulmur's work (https://github.com/raulmur/evaluate_ate_scale) provide scale caculation.

If you add this two functions, your work will be more successful and attract more attention.Thank you very much.

MichaelGrupp, 2019-05-02:
> Could the full trakectory error be saved with the index or timestamp, such as as txt format. the saved data could be used for deep analysis

You can use `--save_results result.zip` to save the full result data, including the raw error values. The timestamps are included if they are available. You have different options: 

1. Use the `evo_res` command line tool.

2. Use this in your code:
```python
from evo.tools import file_interface

result = file_interface.load_res_file("./result.zip")
errors = result.np_arrays["error_array"]
timestamps = result.np_arrays["timestamps"]
```

3. Load the "`.npz`" files inside the `.zip` using [numpy.load()](https://docs.scipy.org/doc/numpy/reference/generated/numpy.load.html).

---

> Raulmur's work (https://github.com/raulmur/evaluate_ate_scale) provided scale caculation.

`evo_ape tum <groundtruth> <trajectory> --align --correct_scale` does a full Sim(3) alignment that includes scale.

cumtchenchang, 2019-05-04:
@MichaelGrupp  Thank you for your responses.
1) for full trajectory error, I followed your suggestion, and got figures such as:
[image]
[image]
However, I can not add legend for violin figure, May you give some suggestions.

MichaelGrupp, 2019-05-06:
Hm, I don't know your code... `plt.legend(...)` if you're using matplotlib?

closed 2019-05-11 by MichaelGrupp