MichaelGrupp/evo #161: Error with TUM trajectory format file

kratochviljakub, 2019-05-02:
When I want to plot the trajectory, which I got from DSO or LDSO (after removing some extra spaces) it says "[ERROR] TUM trajectory files must have 8 entries per row and no trailing delimiter at the end of the rows (space)". But I fixed the file and it has 8 columns and spaces at the right place. The original ground truth from datasets works well.
[result_dso_01.txt](https://github.com/MichaelGrupp/evo/files/3139688/result_dso_01.txt)

MichaelGrupp, 2019-05-02:
Your file is encoded as "UTF-8 with BOM" ([byte order mark](https://en.wikipedia.org/wiki/Byte_order_mark)). Maybe this was added by a text editor, Windows notepad likes to do that sometimes for example. This byte order mark is placed in the first 3 bytes of the file and leads to the crash because it's not expected.

The file can be loaded if the first 3 bytes with the BOM are discarded:
```bash
# (on linux)
tail -c +4 result_dso_01.txt > without-bom.txt
```
[without-bom.txt](https://github.com/MichaelGrupp/evo/files/3140013/without-bom.txt)

---

I will keep this issue open and check if I can make my stuff compatible with this.

kratochviljakub, 2019-05-02:
You're right, even about the editor where I was replacing multiple spaces and it added this mark.

Now it works fine, thank you!

MichaelGrupp, 2019-05-04:
This is now automatically handled since https://github.com/MichaelGrupp/evo/commit/7d636dcd89167e60988104947d28a5c7fb110307

closed 2019-05-04 by MichaelGrupp