MichaelGrupp/evo #163: How to change the color of second traj? Not the reference traj

SALOER, 2019-05-09:


MichaelGrupp, 2019-05-11:
`evo_ape` uses a colormap ("cmap") to display the error values on the trajectory. You can choose another [colormap that works with matplotlib](https://matplotlib.org/users/colormaps.html), e.g.: 

```
evo_config set plot_trajectory_cmap viridis
```
---
`evo_traj` can display multiple trajectories and uses a color cycle for them. The cycle is taken from a [color palette of the seaborn package](https://seaborn.pydata.org/tutorial/color_palettes.html). You can change this to be a different palette:
```
evo_config set plot_seaborn_palette cubehelix
```
or a list of custom colors:
```bash
evo_config set plot_seaborn_palette red blue "#6bb0d6"
```
Custom colors need this fix: #164, so an upgrade to 1.4.3 is needed.

If you need more options you might need to do some customization in the code.

MichaelGrupp, 2019-05-11:
There's also `plot_trajectory_alpha` if you want to change the transparency.

SALOER, 2019-05-11:
> `evo_ape` uses a colormap ("cmap") to display the error values on the trajectory. You can choose another [colormap that works with matplotlib](https://matplotlib.org/users/colormaps.html), e.g.:
> 
> ```
> evo_config set plot_trajectory_cmap viridis
> ```
> 
> `evo_traj` can display multiple trajectories and uses a color cycle for them. The cycle is taken from a [color palette of the seaborn package](https://seaborn.pydata.org/tutorial/color_palettes.html). You can change this to be a different palette:
> 
> ```
> evo_config set plot_seaborn_palette cubehelix
> ```
> 
> or a list of custom colors:
> 
> ```
> evo_config set plot_seaborn_palette red blue "#6bb0d6"
> ```
> 
> Custom colors need this fix: #164, so an upgrade to 1.4.3 is needed.
> 
> If you need more options you might need to do some customization in the code.

Thanks a lot！

closed 2019-05-11 by MichaelGrupp

rocomanmelli, 2020-12-22:
> evo_config set plot_seaborn_palette red blue "#6bb0d6"

I'm afraid that I can't make this work. Only `red` is taken as part of the `plot_seaborn_palette` parameter. Am I missing something? I'm using python 2.7. May be I should open a new issue...