MichaelGrupp/evo #165: ImportError: cannot import name NavigationToolbar2Tk

ajay1606, 2019-05-10:
**Description:**
@MichaelGrupp Thank you for the nice tool for accuracy evaluation. I have tested this package with given test sample data. Unfortunately, I am getting an error when I run my own data.
I have tried running trajectory file with kitty format and ba file format. But still, I am getting error

**Command:**
```evo_traj kitti KITTI_00_ORB.txt KITTI_00_SPTAM.txt --ref=KITTI_00_gt.txt -p --plot_mode=xz
```
evo_traj kitti gnss_pose1.txt ndt_pose.txt -p --plot_mode=xz
**Console output:**
--------------------------------------------------------------------------------
name:	gnss_pose1
infos:	3071 poses, 1366.888m path length
--------------------------------------------------------------------------------
name:	ndt_pose
infos:	3071 poses, 2550.590m path length
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 90, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_traj.py", line 406, in run
    plot_collection.show()
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/plot.py", line 191, in show
    self.tabbed_tk_window()
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/plot.py", line 148, in tabbed_tk_window
    from matplotlib.backends.backend_tkagg import FigureCanvasTkAgg, NavigationToolbar2Tk
ImportError: cannot import name NavigationToolbar2Tk

[DEBUG][2019-05-10 10:47:42,794][main_traj.run():275]
--------------------------------------------------------------------------------
[DEBUG][2019-05-10 10:47:42,845][file_interface.read_kitti_poses_file():161]
Loaded 3071 poses from: gnss_pose1.txt
[DEBUG][2019-05-10 10:47:42,881][file_interface.read_kitti_poses_file():161]
Loaded 3071 poses from: ndt_pose.txt
[INFO][2019-05-10 10:47:42,882][main_traj.print_traj_info():225]
--------------------------------------------------------------------------------
[INFO][2019-05-10 10:47:42,882][main_traj.print_traj_info():226]
name:	gnss_pose1
[INFO][2019-05-10 10:47:42,883][main_traj.print_traj_info():253]
infos:	3071 poses, 1366.888m path length
[INFO][2019-05-10 10:47:42,883][main_traj.print_traj_info():225]
--------------------------------------------------------------------------------
[INFO][2019-05-10 10:47:42,883][main_traj.print_traj_info():226]
name:	ndt_pose
[INFO][2019-05-10 10:47:42,885][main_traj.print_traj_info():253]
infos:	3071 poses, 2550.590m path length
[ERROR][2019-05-10 10:47:44,114][entry_points.launch():99]
Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 90, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_traj.py", line 406, in run
    plot_collection.show()
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/plot.py", line 191, in show
    self.tabbed_tk_window()
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/plot.py", line 148, in tabbed_tk_window
    from matplotlib.backends.backend_tkagg import FigureCanvasTkAgg, NavigationToolbar2Tk
ImportError: cannot import name NavigationToolbar2Tk
[ERROR][2019-05-10 10:47:44,115][entry_points.launch():106]
evo module evo.main_traj crashed - see /home/ajay/.evo/evo.log

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.4.2
* Python version shown by `evo pkg --pyversion`: 2.7.12
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
color
{
    "console_logging_format": "%(message)s", 
    "global_logfile_enabled": true, 
    "plot_backend": "TkAgg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_multi_cmap": "none", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true
}

{
    "console_logging_format": "%(message)s", 
    "global_logfile_enabled": true, 
    "plot_backend": "TkAgg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_multi_cmap": "none", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true
}
Kind regards,
Ajay
[image]

MichaelGrupp, 2019-05-11:
I think you need to update `matplotlib`.

ajay1606, 2019-05-13:
@MichaelGrupp Thank you for your reply. It was still a problem with matplotlib in the python version 2.7. Hence I changed to python version 3.5 and its working well now. Thanks a lot.

closed 2019-05-13 by ajay1606