MichaelGrupp/evo #166: Rigid transform aligment?

leCaTo, 2019-05-17:
Hi, really nice piece of software that you put together here! congrats from the S-PTAM team! We have something quite similar but you completely nailed it.

A feature that we never had time to tackle is an aligner that only uses two rigid transformations for the whole trajectory, one for the inertial frame and one for the base frame. The objective would be to automatically align trajectories for simple comparisons or visual inspection, is also useful as a seed for more "aggressive" alignment like Umeyama's. Of course this won't deal with unfairness of SLAM algorithms like monocular vs stereo scale situation.

ETH's trajectory_toolkit seems to have something on this. [Here](https://github.com/ethz-asl/trajectory_toolkit/blob/master/src/trajectory_toolkit/TimedData.py) they have a way to calibrate those transformations (calibrateInertialTransform and calibrateBodyTransform functions) and apply it to trajectories.

Have you ever though of this?? it might be a nice feature.

MichaelGrupp, 2019-05-24:
Hey, thanks! :)

Just to make sure I understand it correctly (I didn't read the whole code from ETH)... you would like to align the trajectory's origin to the groundtruth's origin?

If that's the case, this can be done with something like this (you can run this in the `test/data` folder of this repository):

```python
from evo.core import lie_algebra as lie
from evo.tools import file_interface

gt = file_interface.read_tum_trajectory_file("./fr2_desk_groundtruth.txt")
traj = file_interface.read_tum_trajectory_file("./fr2_desk_ORB.txt")

traj_to_gt = lie.relative_se3(traj.poses_se3[0], gt.poses_se3[0])
traj.transform(traj_to_gt)

file_interface.write_tum_trajectory_file("./fr2_desk_ORB_transformed.txt", traj)
```

Then I get this:
```
evo_traj tum --ref fr2_desk_groundtruth.txt fr2_desk_ORB.txt fr2_desk_ORB_transformed.txt --plot
```
[image]

leCaTo, 2019-05-24:
Yes, that would be 'kickoff' idea. You can always use first poses to transform inertial frames and align trajectorie's origin. The problem gets trickier if trajectories are tracking different body frames.

The behavior could be extended to support cases where different methods tracks different body frames. An EuRoC inspired example, suppose that you have:
- A ground truth that tracks IMU frame wrt the VICON frame.
- Method 1 that tracks IMU frame wrt the first IMU pose (IMU_0).
- Method 2 that tracks CAM frame wrt the first CAM pose (CAM_0).

Method 1 can be aligned with only a pre-multiplication taking inertial frame IMU_0 to the VICON. But Method 2 would require also a post-multiplication that transforms the tracked body frame CAM to IMU. Otherwise they would start at the same spot but due to axis standards Method 2 will 'go forward' on Z-axis while Method 1 will 'go forward' on X-axis (plus some others rotation peculiarities)

Of course, this can be easily done when knowing methods calibration and configuration, using a user defined post-multiply transformation. The open question is, if there is an automatic way to settle on a pre and post multiplication that would effectively align both inertial and body frames without extra knowledge.

We kind of saw ETH's toolkit as hint that it might be possible but never really got too deep into that calibrateBodyTransform code.

MichaelGrupp, 2019-05-26:
I see, but I'm not sure if such an extrinsic sensor transformation can be recovered accurately enough based only on the trajectories.

> We kind of saw ETH's toolkit as hint that it might be possible but never really got too deep into that calibrateBodyTransform code.

If you can recover a "generic" algorithm from that, we could think about if it's useful to add this also to this repository in some way. I personally probably won't dive deeper into the ETH code in the upcoming time.

On the other side, the extrinsic parameters of the drone are available in the EuRoC dataset, so at least for this dataset one can apply the post-multiplication to the trajectory poses.

---

P.S.: I added an `--align_origin` flag in the latest version that does exactly what the script in the comment above does :wink: Doesn't solve your problem here, but I think it can be useful in general.

MichaelGrupp, 2020-03-30:
Closing for now, because of inactivity

closed 2020-03-30 by MichaelGrupp

JavierTG97, 2021-09-21:
I believe I'm dealing with a similar situation where I want to compare the estimated trajectory of an object whose world reference is created with arbitrariness. This can be exemplified with the following scheme:

[image]

With this set-up, as commented above,  I believe that two rigid transformations are needed to compare the trajectories (in bold):

[image]

@leCaTo @MichaelGrupp Have you come up with a way of dealing with this kind of trajectories? Having a look at the ```--align-origin``` flag, I believe this would not deal with the two transformations needed. Also, the ETH code mentioned above seems to be using one transformation only,

Thanks in advance, and sorry for responding in an old issue
btw, thanks for open-sourcing this amazing project :) It's being very helpful for me.

C-H-Chien, 2024-02-17:
From the document, I did not see a necessary post-transformation step of the EuRoC ground truth dataset before doing evaluations. I thought the code takes care of it before noticing that I was wrong a couple of days later when I came across this discussion thread.  

I suggest that this should be documented so that new users would know how they should properly do evaluations when using the EuRoC dataset.