MichaelGrupp/evo #168: Good align result, but large RPE.

JJQ7, 2019-05-18:
**Description:**
Thanks for providing such useful tool :) 
I met a problem when using evo_rpe on the Euroc dataset. I extract data from rosbag, use evo_traj to align trajectories.  The alignment result is pretty well. However, when using evo_rpe, the given result is out of expectation (which is also larger than the published thesis.) I've tried all the  data bags in Euroc. All of them have the same probems.
Hope to get your advice!
Thanks!

**Command:**
```
evo_rpe tum transform_pose.tum estimated_pose.tum -c ./rpe.json
```

**Console output:**
```
[DEBUG][2019-05-18 18:45:19,095][log.configure_logging():115]
System info:
Python 2.7.12
Linux-4.15.0-45-generic-x86_64-with-Ubuntu-16.04-xenial
quad@quad

[DEBUG][2019-05-18 18:45:19,096][main_rpe.run():228]
main_parser config:
{'align': True,
 'all_pairs': True,
 'config': '../../rpe.json',
 'correct_scale': False,
 'debug': True,
 'delta': 5,
 'delta_tol': 0.1,
 'delta_unit': u'm',
 'est_file': 'estimated_pose.tum',
 'logfile': None,
 'no_warnings': False,
 'plot': False,
 'plot_colormap_max': None,
 'plot_colormap_max_percentile': None,
 'plot_colormap_min': None,
 'plot_full_ref': False,
 'plot_mode': 'xyz',
 'pose_relation': u'trans_part',
 'ref_file': 'transform_pose.tum',
 'save_plot': None,
 'save_results': u'rpe.zip',
 'serialize_plot': None,
 'silent': False,
 'subcommand': 'tum',
 't_max_diff': 0.1,
 't_offset': 0.0,
 'verbose': False}
[DEBUG][2019-05-18 18:45:19,096][main_rpe.run():229]
--------------------------------------------------------------------------------
[DEBUG][2019-05-18 18:45:19,167][file_interface.read_tum_trajectory_file():96]
Loaded 14629 stamps and poses from: transform_pose.tum
[DEBUG][2019-05-18 18:45:19,175][file_interface.read_tum_trajectory_file():96]
Loaded 1392 stamps and poses from: estimated_pose.tum
[DEBUG][2019-05-18 18:45:19,177][main_rpe.run():245]
Synchronizing trajectories...
[DEBUG][2019-05-18 18:45:19,216][sync.associate_trajectories():106]
Found 1392 of max. 14629 possible matching timestamps between...
	transform_pose.tum
and:	estimated_pose.tum
..with max. time diff.: 0.1 (s) and time offset: 0.0 (s).
[DEBUG][2019-05-18 18:45:19,217][main_rpe.rpe():170]
--------------------------------------------------------------------------------
[DEBUG][2019-05-18 18:45:19,217][trajectory.align_trajectory():398]
Aligning using Umeyama's method...
[DEBUG][2019-05-18 18:45:19,223][trajectory.align_trajectory():408]
Rotation of alignment:
[[  1.00000000e+00   2.28766203e-07  -2.59078794e-07]
 [ -2.28766166e-07   1.00000000e+00   1.42894082e-07]
 [  2.59078826e-07  -1.42894023e-07   1.00000000e+00]]
Translation of alignment:
[  7.59104540e-07  -1.96717932e-07  -1.69422631e-07]
[DEBUG][2019-05-18 18:45:19,223][trajectory.align_trajectory():409]
Scale correction: 1.0
[DEBUG][2019-05-18 18:45:19,282][main_rpe.rpe():175]
--------------------------------------------------------------------------------
[DEBUG][2019-05-18 18:45:19,289][filters.id_pairs_from_delta():174]
Found 1242 pairs with delta 5 (m) among 1392 poses using all pairs.
[DEBUG][2019-05-18 18:45:19,333][metrics.process_data():279]
Compared 1242 relative pose pairs, delta = 5 (m) with all pairs.
[DEBUG][2019-05-18 18:45:19,333][metrics.process_data():282]
Calculating RPE for translation part pose relation...
[DEBUG][2019-05-18 18:45:19,338][main_rpe.rpe():193]
--------------------------------------------------------------------------------
[INFO][2019-05-18 18:45:19,338][main_rpe.rpe():194]
RPE w.r.t. translation part (m)
for delta = 5 (m) using all pairs
(with SE(3) Umeyama alignment)

       max	8.828151
      mean	4.289175
    median	4.041543
       min	0.286414
      rmse	4.663418
       sse	27010.353238
       std	1.830422

[DEBUG][2019-05-18 18:45:19,338][main_rpe.run():270]
--------------------------------------------------------------------------------
[DEBUG][2019-05-18 18:45:19,338][file_interface.save_res_file():287]
Saving results to rpe.zip...

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`:  v1.4.1
* Python version shown by `evo pkg --pyversion`:  2.7.12
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04 
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: {
    "console_logging_format": "%(message)s", 
    "global_logfile_enabled": false, 
    "plot_backend": "Qt5Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_multi_cmap": "none", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true
}
```

```

MichaelGrupp, 2019-05-26:
What are the reference frames of the trajectories? You have to be careful with the EuRoC dataset coordinate frames (ground truth, IMU, camera...), because they have different axis conventions.

As far as I remember, the transformations between the frames are explained in the paper and are also available in the dataset.

Umeyama alignment doesn't take this into account because it only aligns points. This is why the alignment looks good in your case (the points match), but the orientation angles don't:
[image]

MichaelGrupp, 2019-05-26:
This comment describes the problem, maybe it helps you: https://github.com/MichaelGrupp/evo/issues/166#issuecomment-495718465

JJQ7, 2019-05-29:
Great thanks! This does really work!
I did not realize this before.

> Umeyama alignment doesn't take this into account because it only aligns points

By the way, I've tried the new feature in V1.5.0 (--align_orign). And it does not perform as expected. I     read the source code
`def align_trajectory_origin(traj, traj_ref):
    """
    align a trajectory's origin to the origin of a reference trajectory
    :param traj: the trajectory to align
    :param traj_ref: reference trajectory
    :return: the aligned trajectory
    """
    if traj.num_poses == 0 or traj_ref.num_poses == 0:
        raise TrajectoryException("can't align an empty trajectory...")
    traj_aligned = copy.deepcopy(traj)
    to_ref_origin = lie.relative_se3(traj.poses_se3[0], traj_ref.poses_se3[0])
    logger.debug("Origin alignment transformation:\n{}".format(to_ref_origin))
    traj_aligned.transform(to_ref_origin)
    return traj_aligned
`
I'm not sure whether only the first pose origin is align in this function. If use more pose to align the orientation, will performance be better?

Thanks a lot!

MichaelGrupp, 2019-06-04:
The origin alignment can be wrong if the coordinate axis conventions don't match, as mentioned. If you think you found another bug, let me know, it's relatively new.

> If use more pose to align the orientation, will performance be better?

Probably, but I'm not planning to implement such a feature at the moment.

closed 2019-06-10 by MichaelGrupp

jingshaojing, 2021-03-23:
> Great thanks! This does really work!
> I did not realize this before.
> 
> > Umeyama alignment doesn't take this into account because it only aligns points
> 
> After transform the whole trajectory, the results seem to be more reasonable.
> 
> By the way, I've tried the new feature in V1.5.0 (--align_orign). And it does not perform as expected. I read the source code
> `def align_trajectory_origin(traj, traj_ref): """ align a trajectory's origin to the origin of a reference trajectory :param traj: the trajectory to align :param traj_ref: reference trajectory :return: the aligned trajectory """ if traj.num_poses == 0 or traj_ref.num_poses == 0: raise TrajectoryException("can't align an empty trajectory...") traj_aligned = copy.deepcopy(traj) to_ref_origin = lie.relative_se3(traj.poses_se3[0], traj_ref.poses_se3[0]) logger.debug("Origin alignment transformation:\n{}".format(to_ref_origin)) traj_aligned.transform(to_ref_origin) return traj_aligned `
> I'm not sure whether only the first pose origin is align in this function. If use more pose to align the orientation, will performance be better?
> 
> Thanks a lot!

I have the same question , how to get right command? thanks! @JJQ7