MichaelGrupp/evo #169: trajectory align problems

goldenminerlmg, 2019-05-24:
**Description:**
I am comparing two trajectories, one is from slam approach and the other is obtained by odometry.  when I use:

**Command:**
```evo_traj tum slam-msf_core-pose.txt --ref=slam-slam_out_pose.txt -p --plot_mode xy

![1](https://user-images.githubusercontent.com/21134987/58329774-ecbc3780-7e67-11e9-94fe-945067bb247e.png)

It show the trajectories well but haven't been aligned. I want to align them so I use this command:

evo_traj tum slam-msf_core-pose.txt --ref=slam-slam_out_pose.txt -p --plot_mode xy --align

![2](https://user-images.githubusercontent.com/21134987/58329797-fc3b8080-7e67-11e9-9c5e-530f5c020e3f.png)

It only show s the trajectory of the reference. Can you give me some advice?
I have use the tail command to check the timestamps of the two dataset, it seems well:
**Command:**
tail slam-slam_out_pose.txt slam-msf_core-pose.txt 
==> slam-slam_out_pose.txt <==
1.558432e+09 -2.662621e+00 -1.459702e+00 0.000000e+00 0.000000e+00 0.000000e+00 -9.959208e-01 9.023139e-02
1.558432e+09 -2.666428e+00 -1.460114e+00 0.000000e+00 0.000000e+00 0.000000e+00 -9.959559e-01 8.984399e-02
1.558432e+09 -2.668152e+00 -1.461731e+00 0.000000e+00 0.000000e+00 0.000000e+00 -9.959364e-01 9.005936e-02
1.558432e+09 -2.661758e+00 -1.459366e+00 0.000000e+00 0.000000e+00 0.000000e+00 -9.959301e-01 9.012941e-02
1.558432e+09 -2.662857e+00 -1.460793e+00 0.000000e+00 0.000000e+00 0.000000e+00 -9.959549e-01 8.985420e-02
1.558432e+09 -2.665939e+00 -1.461960e+00 0.000000e+00 0.000000e+00 0.000000e+00 -9.959446e-01 8.996830e-02
1.558432e+09 -2.664413e+00 -1.459763e+00 0.000000e+00 0.000000e+00 0.000000e+00 -9.959363e-01 9.006007e-02
1.558432e+09 -2.660812e+00 -1.462952e+00 0.000000e+00 0.000000e+00 0.000000e+00 -9.959296e-01 9.013463e-02
1.558432e+09 -2.663223e+00 -1.459106e+00 0.000000e+00 0.000000e+00 0.000000e+00 -9.959331e-01 9.009593e-02
1.558432e+09 -2.662491e+00 -1.460770e+00 0.000000e+00 0.000000e+00 0.000000e+00 -9.959147e-01 9.029882e-02

==> slam-msf_core-pose.txt <==
1.558432e+09 2.680811e+00 -3.013807e+00 3.416595e-01 1.115940e-03 4.968753e-04 7.110882e-01 -7.031018e-01
1.558432e+09 2.679516e+00 -3.010908e+00 3.412600e-01 1.141422e-03 5.614319e-04 7.114423e-01 -7.027434e-01
1.558432e+09 2.679629e+00 -3.011635e+00 3.412282e-01 1.139452e-03 5.691526e-04 7.114434e-01 -7.027423e-01
1.558432e+09 2.679708e+00 -3.012146e+00 3.412057e-01 1.143464e-03 5.672224e-04 7.114472e-01 -7.027384e-01
1.558432e+09 2.679796e+00 -3.012712e+00 3.411808e-01 1.147778e-03 5.616186e-04 7.114539e-01 -7.027316e-01
1.558432e+09 2.678530e+00 -3.009852e+00 3.407761e-01 1.181276e-03 6.179679e-04 7.118117e-01 -7.023691e-01
1.558432e+09 2.678617e+00 -3.010430e+00 3.407497e-01 1.180889e-03 6.153377e-04 7.118152e-01 -7.023656e-01
1.558432e+09 2.678705e+00 -3.011021e+00 3.407228e-01 1.178902e-03 6.106520e-04 7.118158e-01 -7.023650e-01
1.558432e+09 2.678792e+00 -3.011607e+00 3.406960e-01 1.181092e-03 6.109935e-04 7.118178e-01 -7.023629e-01
1.558432e+09 2.677553e+00 -3.008772e+00 3.402872e-01 1.214064e-03 6.713473e-04 7.121664e-01 -7.020093e-01

MichaelGrupp, 2019-05-24:
`--n_to_align N` means that point-to-point alignment for the first `N ` poses is done with respect to the reference trajectory.

It can happen that a low number of `N` is not "stable" enough to do a good alignment. This depends on the number of poses in your trajectories and the shape of the path. Please also keep in mind that the Umeyama method that is used for alignment is only minimizing the distance between the `N` points and does not take orientation of the poses into account. So `N=1` will start at the same point, but the orientation can be compeletely wrong.

goldenminerlmg, 2019-05-25:
So that means if the dataset have two trajectories that with different frequency,  we can't use the --n_to_align N directly. This is why when I compare with them, one of the trajectory seems to deformation. Is this parameter only can be used in the case of the trajectories to be compared are with same frequency?

MichaelGrupp, 2019-05-26:
You can use it, but you might have to adapt it for individual cases. But in general, dealing with different frequencies can be tricky and some custom pre-processing (upsampling, interpolation, ...) could be an option.

MichaelGrupp, 2019-05-26:
This might be also interesting for you: I added a new flag `--align_origin`, this aligns the first pose but also takes rotation into account. https://github.com/MichaelGrupp/evo/commit/4c04630707a4e8e4cf5cd47b45af8c1702f0db93

Available in v1.5.0.

closed 2019-05-26 by MichaelGrupp