MichaelGrupp/evo #170: Problem with Positional Error evaluation from bag file

mnik17, 2019-05-24:
**Description:**
Hello,

first of all thanks for creating this incredibly useful tool. 

I'm trying to evaluate the APE/RPE of [A-LOAM](https://github.com/HKUST-Aerial-Robotics/A-LOAM) using the [KITTI Odometry Dataset](http://www.cvlibs.net/datasets/kitti/eval_odometry.php). What I did was, I ran the algorithm and then recorded the reference (odometry_gt) and estimation (aft_mapped_to_init) in a bag file using the `rosbag record` command. When I try `evo_traj` on this file everything works flawlessly, however when I try using `evo_ape` it tells me that no messages for the topics are recorded in the bag. Note that both topics are  `nav_msgs/Odometry`.

P.S. I could also provide the trajectory plot if necessary.

**Command:**
```
evo_traj bag 2019-05-24-14-20-03.bag --full_check --all_topics -p
```

**Console output:**
```

name:	odometry_gt
infos:
	duration (s)	469.752184093
	nr. of poses	4533
	path length (m)	3717.30752139
	pos_end (m)	[96.96153259  5.58393097  3.56275797]
	pos_start (m)	[6.86547709 0.3749547  0.227029  ]
	t_end (s)	470.581604004
	t_start (s)	0.829419911
checks:
	SE(3) conform	yes
	array shapes	ok
	nr. of stamps	ok
	quaternions	ok
	timestamps	ok
stats:
	v_avg (km/h)	28.488511
	v_avg (m/s)	7.913475
	v_max (km/h)	46.443120
	v_max (m/s)	12.900867
	v_min (km/h)	0.064283
	v_min (m/s)	0.017856
--------------------------------------------------------------------------------
name:	aft_mapped_to_init
infos:
	duration (s)	469.959559202
	nr. of poses	3431
	path length (m)	3718.24313244
	pos_end (m)	[93.98707402 12.91776977 -3.20519212]
	pos_start (m)	[4.36527802 0.09576837 0.04299688]
	t_end (s)	470.581604004
	t_start (s)	0.622044802
checks:
	SE(3) conform	yes
	array shapes	ok
	nr. of stamps	ok
	quaternions	ok
	timestamps	ok
stats:
	v_avg (km/h)	28.348800
	v_avg (m/s)	7.874667
	v_max (km/h)	46.936832
	v_max (m/s)	13.038009
	v_min (km/h)	0.084230
	v_min (m/s)	0.023397

```

zzz

**Command:**
```
evo_ape bag 2019-05-24-14-20-03.bag odometry_gt aft_mapped_to_init
```

**Console output:**
```

no messages for topic 'odometry_gt' in bag
```

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.4.3
* Python version shown by `evo pkg --pyversion`: 2.7.12
* operating system and version (e.g. Ubuntu 16.04 or Windows 10):  Ubuntu 16.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s", 
    "global_logfile_enabled": false, 
    "plot_backend": "Qt5Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_multi_cmap": "none", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true
}

```

MichaelGrupp, 2019-05-24:
Just a wild guess: what is the "official" name of the topic when you run `rosbag info 2019-05-24-14-20-03.bag`? I think you are missing the leading `/` of the topic.

---

`evo_traj` strips away the leading slash of ROS topics before printing (`/bla` is shown as `bla`) [here](https://github.com/MichaelGrupp/evo/blob/master/evo/main_traj.py#L226).

This is useful for other modes where the trajectory name is inside a file path... so that for example:
```
evo_traj tum /some/super/long/path/that/nobody/cares/about/my_trajectory.txt
```
will only show `my_trajectory`.

But for ROS topics this is probably just confusing.

closed 2019-05-26 by MichaelGrupp

ObscureFox, 2020-05-16:
This really help a lot, I find the topic has an extra '/ ' in my bag.     0.0

arunceria, 2020-06-23:
Good day
The meters do not coincide with those given by the tool. Help