MichaelGrupp/evo #171: Unhandled error in evo.main_traj

why-freedom, 2019-05-25:
**Description:**
Ubuntu 16.04 
python2.7 python3.5
When running the Evo times wrong, I follow your tutorial to install, but the error,,, thank you for your help

**Command:**
```why@why-desktop:~/Documents/SLAM_dateset/evo/test/data$ evo_traj kitti KITTI_00_ORB.txt KITTI_00_SPTAM.txt --ref=KITTI_00_gt.txt -p --plot_mode=xz
# replace this line with the command(s) you used
```

**Console output:**
```--------------------------------------------------------------------------------
name:	KITTI_00_ORB
infos:	4541 poses, 3705.098m path length
--------------------------------------------------------------------------------
name:	KITTI_00_SPTAM
infos:	4541 poses, 3718.501m path length
--------------------------------------------------------------------------------
name:	KITTI_00_gt
infos:	4541 poses, 3724.187m path length
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/home/why/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/entry_points.py", line 90, in launch
    main_module.run(args)
  File "/home/why/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/main_traj.py", line 342, in run
    from evo.tools.plot import PlotMode
  File "/home/why/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/tools/plot.py", line 35, in <module>
    import matplotlib.pyplot as plt
  File "/home/why/.virtualenvs/evaluation/local/lib/python2.7/site-packages/matplotlib/pyplot.py", line 115, in <module>
    _backend_mod, new_figure_manager, draw_if_interactive, _show = pylab_setup()
  File "/home/why/.virtualenvs/evaluation/local/lib/python2.7/site-packages/matplotlib/backends/__init__.py", line 63, in pylab_setup
    [backend_name], 0)
  File "/home/why/.virtualenvs/evaluation/local/lib/python2.7/site-packages/matplotlib/backends/backend_qt5agg.py", line 15, in <module>
    from .backend_qt5 import (
  File "/home/why/.virtualenvs/evaluation/local/lib/python2.7/site-packages/matplotlib/backends/backend_qt5.py", line 19, in <module>
    import matplotlib.backends.qt_editor.figureoptions as figureoptions
  File "/home/why/.virtualenvs/evaluation/local/lib/python2.7/site-packages/matplotlib/backends/qt_editor/figureoptions.py", line 20, in <module>
    import matplotlib.backends.qt_editor.formlayout as formlayout
  File "/home/why/.virtualenvs/evaluation/local/lib/python2.7/site-packages/matplotlib/backends/qt_editor/formlayout.py", line 54, in <module>
    from matplotlib.backends.qt_compat import QtGui, QtWidgets, QtCore
  File "/home/why/.virtualenvs/evaluation/local/lib/python2.7/site-packages/matplotlib/backends/qt_compat.py", line 168, in <module>
    raise ImportError("Failed to import any qt binding")
ImportError: Failed to import any qt binding

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```

MichaelGrupp, 2019-05-26:
Looks like you need to install PyQt5 before you can use the Qt5 backend. See here: 
https://stackoverflow.com/questions/52346254/importerror-failed-to-import-any-qt-binding-python-tensorflow

Or, you don't use it and use the default TkAgg backend instead: `evo_config set plot_backend tkagg`

why-freedom, 2019-05-27:
Sorry to bother you. I have configured the PyQt5, matplotlib, numpy and other libraries, but when I try to draw the TUM track, I still have the following problems, I hope to get your help, I am grateful.Thanks...

why@why-desktop:~/Documents/SLAM_dateset/evo/test/data$ evo_traj tum fr2_desk_ORB.txt -p --plot_mode=xy
--------------------------------------------------------------------------------
name:	fr2_desk_ORB
infos:	2893 poses, 21.730m path length, 98.822s duration
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/evo/entry_points.py", line 90, in launch
    main_module.run(args)
  File "/usr/local/lib/python3.5/dist-packages/evo/main_traj.py", line 342, in run
    from evo.tools.plot import PlotMode
  File "/usr/local/lib/python3.5/dist-packages/evo/tools/plot.py", line 41, in <module>
    import seaborn as sns
  File "/home/why/.local/lib/python3.5/site-packages/seaborn/__init__.py", line 17, in <module>
    from .widgets import *
  File "/home/why/.local/lib/python3.5/site-packages/seaborn/widgets.py", line 8, in <module>
    from ipywidgets import interact, FloatSlider, IntSlider
  File "/usr/local/lib/python3.5/dist-packages/ipywidgets/__init__.py", line 25, in <module>
    from .widgets import *
  File "/usr/local/lib/python3.5/dist-packages/ipywidgets/widgets/__init__.py", line 4, in <module>
    from .widget import Widget, CallbackDispatcher, register, widget_serialization
  File "/usr/local/lib/python3.5/dist-packages/ipywidgets/widgets/widget.py", line 15, in <module>
    from ipykernel.comm import Comm
  File "/usr/local/lib/python3.5/dist-packages/ipykernel/comm/__init__.py", line 1, in <module>
    from .manager import *
  File "/usr/local/lib/python3.5/dist-packages/ipykernel/comm/manager.py", line 15, in <module>
    from .comm import Comm
  File "/usr/local/lib/python3.5/dist-packages/ipykernel/comm/comm.py", line 9, in <module>
    from ipykernel.kernelbase import Kernel
  File "/usr/local/lib/python3.5/dist-packages/ipykernel/kernelbase.py", line 19, in <module>
    from jupyter_client.session import utcnow as now
  File "/usr/local/lib/python3.5/dist-packages/jupyter_client/session.py", line 56, in <module>
    from zmq.eventloop.ioloop import IOLoop
  File "/usr/local/lib/python3.5/dist-packages/zmq/eventloop/__init__.py", line 3, in <module>
    from zmq.eventloop.ioloop import IOLoop
  File "/usr/local/lib/python3.5/dist-packages/zmq/eventloop/ioloop.py", line 21, in <module>
    from tornado import ioloop
  File "/usr/local/lib/python3.5/dist-packages/tornado/ioloop.py", line 45, in <module>
    from tornado.concurrent import (
  File "/usr/local/lib/python3.5/dist-packages/tornado/concurrent.py", line 175, in <module>
    future: Union["futures.Future[_T]", "Future[_T]"], value: _T
  File "/usr/lib/python3.5/typing.py", line 552, in __getitem__
    dict(self.__dict__), parameters, _root=True)
  File "/usr/lib/python3.5/typing.py", line 512, in __new__
    for t2 in all_params - {t1} if not isinstance(t2, TypeVar)):
  File "/usr/lib/python3.5/typing.py", line 512, in <genexpr>
    for t2 in all_params - {t1} if not isinstance(t2, TypeVar)):
  File "/usr/lib/python3.5/typing.py", line 190, in __subclasscheck__
    self._eval_type(globalns, localns)
  File "/usr/lib/python3.5/typing.py", line 177, in _eval_type
    eval(self.__forward_code__, globalns, localns),
  File "<string>", line 1, in <module>
TypeError: 'type' object is not subscriptable

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)

why-freedom, 2019-05-27:
I have solved this problem,thank you.
https://github.com/jupyter/notebook/issues/4439#issue-416244445

why-freedom, 2019-05-27:
> I have solved this problem,thank you.
> [jupyter/notebook#4439 (comment)](https://github.com/jupyter/notebook/issues/4439#issue-416244445)

closed 2019-05-27 by why-freedom

MichaelGrupp, 2019-05-27:
Ok😊