MichaelGrupp/evo #175: question about question

GentlemanCheng, 2019-05-29:
hi, I was reminded that the result.txt of dso slam by monocular dataset are not tum format when I used evo to estimated the result. But when I reviewed the code for dso, I found the result.txt should be tum format, so I am confused. If it is not the tum format, how can I saved the result.txt as tum format ? Thank you!

MichaelGrupp, 2019-05-29:
Can you upload that file here, please?

GentlemanCheng, 2019-05-30:
Ok, I have uploaded the result.txt of dso_slam and the groundtruthSync.txt from monocular_dataset, and this is corresponding code link https://github.com/JakobEngel/dso/blob/ae1d0b3d972367b0a384e587852bedcf5f414e69/src/FullSystem/FullSystem.cpp#L260, thanks for your help very much!
[groundtruthSync.txt](https://github.com/MichaelGrupp/evo/files/3235405/groundtruthSync.txt)
[result.txt](https://github.com/MichaelGrupp/evo/files/3235397/result.txt)

MichaelGrupp, 2019-05-30:
Ok... your files are a bit broken:

`result.txt` sometimes has double spaces between the numbers, this leads to the error because only single spaces are expected. The DSO code you linked seems to be fine though.

Removing the double spaces fixes the file. Quick way to do this with UNIX:
```
cat result.txt | tr -s ' ' > clean.txt
```

On the other hand, `groundtruthSync.txt` contains a lot of lines with "NaN" values They can be removed too, but the question is why they are there
```
grep -v "NaN" groundtruthSync.txt > clean_groundtruth.txt
```

GentlemanCheng, 2019-05-31:
Wow, that woks finely. Thank you! I also have no idea what the value "NAN" means, and maybe pose which is not catched. Thanks again for your kind help!

closed 2019-05-31 by MichaelGrupp

Kallista, 2019-06-21:
Regarding the NANs, in the TUM mono dataset the ground truth is only provided for parts of the beginning and end of the sequences. So you can only test absolute drift/loop closure.