MichaelGrupp/evo #180: AttributeError: 'FigureCanvasTkAgg' object has no attribute 'manager'

LongruiDong, 2019-06-25:
**Description:**
After I get the plot result, I want to save the plot as an image format. But when I click  "save button" on the Trajectory plot GUI window, terminal return an ERROR:
```
Exception in Tkinter callback
Traceback (most recent call last):
  File "/home/*/anaconda3/lib/python3.6/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/home/*/anaconda3/lib/python3.6/site-packages/matplotlib/backends/_backend_tk.py", line 688, in save_figure
    master=self.canvas.manager.window,
AttributeError: 'FigureCanvasTkAgg' object has no attribute 'manager'

```
Can you help me with it ?
Thanks a lot !

**Command:**
```
evo_traj kitti /home/*/catkin_ws/src/A-LOAM/result/00rwc.txt --ref=/home/*/kitti-benchmark/dataset/poses/00.txt -p --plot_mode=xz -a -s --align_origin 

```

**Console output:**
```

--------------------------------------------------------------------------------
name:	00rwc
infos:	4541 poses, 1782.711m path length
--------------------------------------------------------------------------------
name:	00
infos:	4541 poses, 3724.187m path length
Exception in Tkinter callback
Traceback (most recent call last):
  File "/home/*/anaconda3/lib/python3.6/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/home/*/anaconda3/lib/python3.6/site-packages/matplotlib/backends/_backend_tk.py", line 688, in save_figure
    master=self.canvas.manager.window,
AttributeError: 'FigureCanvasTkAgg' object has no attribute 'manager'

```

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`:  
* operating system and version (e.g. Ubuntu 16.04 or Windows 10):  
* did you change the source code? (yes / no):  
* output of `evo_config show --brief --no_color`: 

```
v1.5.1
3.6.8
Ubuntu 16.04
no
{
    "console_logging_format": "%(message)s",
    "global_logfile_enabled": false,
    "plot_backend": "TkAgg",
    "plot_export_format": "pdf",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_multi_cmap": "none",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true
}

```

MichaelGrupp, 2019-06-25:
Looks like a bug in matplotlib that will be fixed in the next version of it: https://github.com/matplotlib/matplotlib/issues/14602

I couldn't reproduce your problem with Python 3.5.2, matplotlib 3.0.3 and python3-tk 3.5.1-1

MichaelGrupp, 2019-06-25:
You can use `--save_figure /tmp/plot.pdf` as a workaround and adjust the default plot settings using `evo_config` if needed: https://github.com/MichaelGrupp/evo/wiki/Plotting

nchauvin74, 2019-07-02:
I get similar errors on mouse motion events with a Matplotlib plot on a new Tkinter window ( 
Toplevel(root) ) when trying to use Matplotlib NavigationToolbar to zoom on the plot.

I get no error when moving the mouse on the plot before using the navigation toolbar.
Once an icon of the navigation toolbar is selected, these errors appear.
If the icon is un-clicked, the errors disappear.

Configuration:
    Windows 7 64-bit
    Python 3.7.1
    Matplotlib  3.1.0
    tkinter.TkVersion 8.6

Code:
        window2 = Toplevel(root)
        fig2 = plt.figure()
        ax2 = plt.subplot(1, 1, 1)
        canvas2 = FigureCanvasTkAgg(fig2, master=window2)
        canvas2.get_tk_widget().grid(row=1,column=4,columnspan=3,rowspan=20)
        ax2.plot(x, y, color=(0.0, 0.0, 1.0))
        toolbarFrame2 = Frame(master=window2)
        toolbarFrame2.grid(row=22,column=4)
        toolbar2 = NavigationToolbar2Tk(canvas2, toolbarFrame2)
        #canvas2.mpl_connect('motion_notify_event', on_mouse_move_event)

Error:
    Traceback (most recent call last):
      File "C:\Users\srv-corp-ost\AppData\Local\Programs\Python\Python37-32\lib\site-packages\matplotlib\cbook\__init__.py", line 216, in process
        func(*args, **kwargs)
      File "C:\Users\srv-corp-ost\AppData\Local\Programs\Python\Python37-32\lib\site-packages\matplotlib\backend_bases.py", line 2708, in mouse_move
        self._set_cursor(event)
      File "C:\Users\srv-corp-ost\AppData\Local\Programs\Python\Python37-32\lib\site-packages\matplotlib\backend_bases.py", line 2700, in _set_cursor
        self.set_cursor(cursors.SELECT_REGION)
      File "C:\Users\srv-corp-ost\AppData\Local\Programs\Python\Python37-32\lib\site-packages\matplotlib\backends\_backend_tk.py", line 613, in set_cursor
        window = self.canvas.manager.window
    AttributeError: 'FigureCanvasTkAgg' object has no attribute 'manager'

MichaelGrupp, 2019-07-02:
Closing because this looks like an issue in matplotlib, so please head to the matplotlib issue tracker if the fix that I linked above doesn't help.

closed 2019-07-02 by MichaelGrupp