MichaelGrupp/evo #181: KITTI dataset strange rotation results

giwrgosg, 2019-06-26:
I would like to ask if anyone can translate the rotation results for the KITTI dataset. I get the following:

[image]

Yaw is fine, however I can't understand why pitch and roll get such high values. The thing is that they get these values even in the provided groundtruth data.

MichaelGrupp, 2019-06-27:
I'm not sure what you mean exactly, but take into account that the axis is normalized to [-180, 180] degrees. So sign flips can happen when the values are close to 180 degrees.

giwrgosg, 2019-06-27:
Yes I know that they are normalized, however if pitch is 90 deg as it shows, then the car should point to the sky, also roll has very high values and the car does not do roll at all, am I misunderstanding something?

MichaelGrupp, 2019-06-27:
Dunno, which sequence is this?

mnik17, 2019-06-28:
That's sequence 00, I've been having the same issues. 

[image]

[image]

For that plot I've used the ground truth data as provided from KITTI, that means in their camera coordinate system.  That's why the plot is rotated. It also means that when the car turns left or right, the change is reflected in the pitch, since the y-axis is pointing to the ground. So because the car obviously moves on a relatively flat road it's obviously not possible for the roll and yaw angles to spike so steeply and not to this extent.

However following the sequnce trajectory, even the pitch is not entirely correct. I sadly don't have the plot right now, but I'll try to explain. The car makes right turn(pitch +90 deg), then a left (0 deg), left (-90 deg), right (0 deg), left (-90 deg). So far it's completely true. Then it makes another two lefts and the pitch goes to 0 and then to 90 degrees, in other words starts turning in the wrong direction. 

Interestingly enough while I was trying to transform the ground truth data to the lidar coordinate system I got very interesting results: 

[image]
[image]

For the transformation I multiplied the poses with two rotational matrices in order to rotate the axes. First of all I rotated by 90 degrees w.r.t. y-axis and then by -90 degrees w.r.t x-axis. 
The rotation matrices that I used:
R_y:               R_x:
0  0  1            1  0  0
0  1  0            0  0  1
-1 0  0            0  -1 0
 
As you can see in the pictures, after the transformation the roll all angles have reasonable metrics (bear in mind the angles don't correspond exactly to the ones in the first pictures, because of the rotation). However I can't explain why the amplitude were affected when I just rotated the trajectory.

MichaelGrupp, 2019-06-29:
> roll, pitch and yaw relevant for the APE, RPE?

In this code RPY angles are only used for this one plot in `evo_traj`.

Rotation in general is relevant depending on your choice of `--pose_relation`. The math is described briefly here: https://github.com/MichaelGrupp/evo/blob/master/notebooks/metrics.py_API_Documentation.ipynb

MichaelGrupp, 2019-07-02:
So it looks like the problem is that the names "roll pitch yaw" interpretation is not what you expect here.

As I said, here I also just use it for this one plot, using an [sxyz rotation order by default](https://github.com/MichaelGrupp/evo/blob/e6ad82c9700a406aeeec1b1f5216ae177eb3bd82/evo/core/trajectory.py#L114) as specified by the [transformations.py](https://github.com/MichaelGrupp/evo/blob/master/evo/core/transformations.py#L87) module. This might not necessarily be the right order for any case. In the KITTI case of a camera frame mounted on a car, szxy makes probably more sense, then the plot looks like this:
[image]

---

**But**: Besides visualization I wouldn't care too much about RPY for anything serious, it's a very weak 3d rotation representation for various reasons.

Roshnee, 2019-07-02:
> So it looks like the problem is that the names "roll pitch yaw" interpretation is not what you expect here.
> 
> As I said, here I also just use it for this one plot. Thi is done using a more or less arbitrarily chosen [sxyz rotation order by default](https://github.com/MichaelGrupp/evo/blob/e6ad82c9700a406aeeec1b1f5216ae177eb3bd82/evo/core/trajectory.py#L114) as specified by the [transformations.py](https://github.com/MichaelGrupp/evo/blob/master/evo/core/transformations.py#L87) module. This might not necessarily be the right order for any case. In the KITTI case of a camera frame mounted on a car, szxy makes probably more sense, then the plot looks like this:
> [image]
> 
> **But**: Besides visualization I wouldn't care too much about RPY for anything serious, it's a very weak 3d rotation representation for various reasons.

Alright!! This makes sense now. Then that has to be changed in the code. I have installed a non-editable version with only executables. Can i install an editable version using 
`pip install --editable . --upgrade --no-binary evo` on top of it?

MichaelGrupp, 2019-07-02:
> Can i install an editable version using pip install --editable . --upgrade --no-binary evo on top of it?

It should just replace the installed one.

> But the values that it takes and the comparison of results were affected i guess??

Nope. Rotational error calculation is **not** based on RPY, see the link to the math above or the code.

Roshnee, 2019-07-02:
Okay, I will try it. Thank you!!

closed 2019-07-03 by MichaelGrupp

YznMur, 2022-11-29:
Hi @MichaelGrupp @Roshnee 
I have the same problem with rotation angles for KITTI here. I changed the axis as you mentioned earlier to SZXY but it still doesn't work (the values are not real). cloud you please tell me how did you manage to fix that and where to modify in the code? thanks.

[image]

MichaelGrupp, 2022-11-29:
FYI: This was added as a configuration option, no change in the code needed anymore.

```
evo_config set euler_angle_sequence szxy
```

```
evo_config show | grep euler -A2
```

PR #186 (2019-07-03): Configurable Euler axes order for evo_traj's plot.
Closes #185 and closes #181.