MichaelGrupp/evo #182: install the evo success, when i tip evo_ape , it couldn't work. it means i cann't find the executal command

jyzhong, 2019-06-28:


MichaelGrupp, 2019-06-29:
Please provide more details. Apart from that, you can check if `pip show -f evo` shows you the location of installed files,

jyzhong, 2019-07-01:
when i use  pip show -f evo, it shows:
Name: evo
Version:1.5.1
.....
.....
Files:
Cannot locate installed-files.txt
BTW: i can use main_rpe replace evo_ape.

jyzhong, 2019-07-01:
> Please provide more details. Apart from that, you can check if `pip show -f evo` shows you the location of installed files,

when i use pip show -f evo, it shows:
Name: evo
Version:1.5.1
.....
.....
Files:
Cannot locate installed-files.txt
BTW: i can use main_rpe replace evo_ape.

MichaelGrupp, 2019-07-04:
* operating system?
* virtualenv?
* which command do you use to install?

jyzhong, 2019-07-04:
>     * operating system?
> 
>     * virtualenv?
> 
>     * which command do you use to install?

ubuntu 18.04.02 LTS,
I install it by pip and source. Both show install success. First i install by pip,  couldn't use the evo_xxx; then i install by source, the answer is the same. Actually, i use this useful tool more than once, but now it doesn't work. Maybe "Cannot locate installed-files.txt" contain some messages.

jyzhong, 2019-07-04:
>     * operating system?
> 
>     * virtualenv?
> 
>     * which command do you use to install?

i try again, success.
i upgrade the pip to 19
then evo_xxx worked.
thanks for your reply ! I appreciate it.

closed 2019-07-04 by jyzhong