MichaelGrupp/evo #183: Comparing results from google cartographer

mnik17, 2019-06-28:
Hi @MichaelGrupp, 

I'm not sure if I should post this here or in the Google Cartographer repo, so sorry if I got it wrong. I decided to ask here, because it's kind of related to both, but I also saw that you're active on there as well. 

So I used [this package](https://github.com/inkyusa/cartographer_kitti_config) to get cartographer working with KITTI and the next thing I wanted to do was evaluate the RPE and APE of the calculated trajectory against the ground truth data. I followed [this issue](https://github.com/googlecartographer/cartographer_ros/pull/1099) and I got the trajectory plot out of cartographer in form of a bag file with `geometry_msgs/PoseStamped`.  

My question is, in what format should the ground truth data be in order to be comparable to the output that I got from google cartographer. Since I had the ground truth data recorded in a bag from before that in `nav_msgs/Odometry.msg`, I tried if I could use that and merge the bags, but evo threw an error that there are no matching timestamps between the topics. Is it because the message are of different type, or should indeed both topics be recorded at the same time in order to be evaluated. I could also transform the cartographer trajectory to KITTI txt format somehow, but I don't know if that would be useful, because there are a lot more poses published by cartographer than there are ground truth poses.

Cheers

MichaelGrupp, 2019-06-29:
Timestamps should match, or at least the offset should be compensated if you know it exactly (`--t_offset`).

I don't know how the timestamps of your KITTI data bagfiles look like, but I would activate [ROS simulation time](http://wiki.ros.org/Clock#Using_Simulation_Time_from_the_.2BAC8-clock_Topic) and play the dataset bag with `--clock`, so you can reuse the time of the dataset exactly.

mnik17, 2019-07-03:
Sorry for answering a bit late, but I got to this issue just now. 

Synchronising timestamps with the sim time option certainly solved the problem and I can run the metrics on the bags. 

Thanks for the help.

closed 2019-07-03 by mnik17

ashnarayan13, 2019-07-06:
Hi @mnik17 ,

I am recording the bagfile with the following launch file. I have two nodes which publish the odometry from cartographer and one that publishes the 

<launch>

<param name="/use_sim_time" value="true" />
<node pkg="eval" name="slam_odometry" type="slam_odometry.py" output="screen"> 
</node>
<node pkg="eval" name="ox_odometry" type="ox_odometry.py" output="screen"> 
</node>
<node pkg="rosbag" type="record" name="my_rosbag" output="screen" args="/stateestimation/OXodometry /stateestimation/slamodometry -o /home/nomad/Datasets/test.bag"/>
</launch>

When I run cartographer with sim_time on as well. I still get the error [ERROR] found no matching timestamps between reference and /stateestimation/slamodometry with max. time diff 0.01 (s) and time offset 0.0 (s)

mnik17, 2019-07-08:
@ashnarayan13 
I'm not sure what your problem is, but you should make sure that the reference topic also has the same timestamps.

SantiagoOrdonez, 2020-03-05:
Hey @mnik17, 

So I am trying to do the same thing you did. I was able to create a bag file from the PoseStamped topic on cartographer but now I am not sure where I am supposed to get the ground truth data for this bag file kitti_2011_09_30_drive_0027_synced_imu_velo.bag? This is the bag file that is in https://github.com/inkyusa/cartographer_kitti_config

Also, how can I make the gt into a bag file?
Any help is appreciated :)

mnik17, 2020-03-05:
Hey @SantiagoOrdonez,

On the KITTI website there is ground truth data for the sequences from 0 to 10. For all other sequences there is no ground truth data provided. I don't remember if they came with GPS data though. If you can find the GPS data for the sequence you can write a simple script to transform in the ground truth data format and coordinate system, since the gt is derived from GPS.

The way I saved the gt in a bag was via ROS. I basically played it in the given format and then I just recorded the topic as a bag file using ROS. Unfortunately I don't remember everything that I did, so I can't give you any specific details on how I achieved it, however I think it was pretty straightforward.

Cheers

mhaboali, 2020-04-21:
Hey @mnik17  @SantiagoOrdonez 

I'm doing the same process now but I don't know how I can get the Poses data from Cartographer. Could you mention the steps(configurations) of how did you achieve that?

Your comments were much helpful, and I'm looking forward to hearing from you soon!
Thanks!

mnik17, 2020-04-22:
Hi @mhaboali,

sadly I can't remember exactly which topic from cartographer I used to get the data from cartographer and I don't have the files any more so I can't check either. You can either try to run cartographer and list all active topics and then just check their data or if you still can't find it, then maybe someone in the cartographer repo could give you more info.

mhaboali, 2020-04-22:
Thanks @mnik17 for your reply, I appreciate that!

Yeah I couldn't find Pose*, Path, or Odometry data with the active topics. And I didn't get from their documentation how to publish it. 

I generated .pbstream file hoping that would let me extract pose information from it but I wasn't able to do either :/

MichaelGrupp, 2020-04-22:
Depending on what you want:

* record `/tf` topic for live data
* use `/write_state` service to serialize the pose graph and then `cartographer_ros cartographer_dev_pbstream_trajectories_to_rosbag`
* `/trajectory_query` service to get a single trajectory from the pose graph

mhaboali, 2020-04-22:
Thanks so much @MichaelGrupp!

I'll try it in a while and will keep you updated if it works for me.

I really appreciate your great collaboration!

mnik17, 2020-04-22:
@mhaboali 

@MichaelGrupp is right. Now that he mentioned it, I'm pretty sure I used the built-in cartographer function, which dumps the pbstream as a rosbag. 

Since I saw you also posted a question regarding HDL Graph SLAM I can also tell you that it also has an option to dump the generated graph data as some sort of a text file, which you can then parse and get the info from. Basically it will contain information about all nodes and egdes.

But as I said I can't really help you further than that because I've been working on completely different things since so I can't remember all the details.

mhaboali, 2020-04-22:
Thanks @mnik17, I appreciate your effort to help!

I'll update the issue when I get a chance to get things working.