MichaelGrupp/evo #184: Coordinate system used to represent output

Roshnee, 2019-07-02:
Hello,

I have been using evo to plot trajectories for kitti dataset. I suppose that the output is represented in the camera coordinate system (X points rightwards, Y points downwards and Z points forwards) - from the kitti paper. When i plot in the XZ plane it it gives me the top view and when I plot in XY plane it gives me the front view. I would like to know if it matters? Does the representation have to be in global coordinate system (Z upward, Y leftward and X forward)?  Which coordinate system is being used for representation? Am i correct or have I understood something in a wrong way? 

Please help. Thanks in advance

MichaelGrupp, 2019-07-02:
It is up to you to set the coordinate axis convention. It should be consistent though if you use/compare multiple trajectories.

Roshnee, 2019-07-02:
Got it! Thank you for your immediate response

closed 2019-07-02 by Roshnee