MichaelGrupp/evo #188: about kitti ape

xiaozhi12345678, 2019-07-13:
Hi，I am confused that you say kitti dataset cant evaluate ape because of it does not have timestamps，is there any method to evaluate it‘s ape？thank you！

MichaelGrupp, 2019-07-13:
Where did I say what? You just need to make sure ground truth and estimated trajectory have the exact same number of corresponding poses (if using the "KITTI" format without timestamps).

When this is fulfilled, you can use `evo_ape kitti`, as it's also shown in the Readme example.

xiaozhi12345678, 2019-07-14:
emmm，I record the key frame poses,so the number of poses is not the same as ground truth,is there any method to solve the problem?thank you.

MichaelGrupp, 2019-07-14:
Then you need to have timestamps. See here for a similar problem: https://github.com/MichaelGrupp/evo/issues/152#issuecomment-473660536

xiaozhi12345678, 2019-07-15:
oh,I got it,Thanks!I have another question ,which command could compute rotation error?I input "evo_ape kitti *.txt *.txt",it printf "APE w.r.t. translation part" and I want rotation part.

MichaelGrupp, 2019-07-15:
`--pose_relation angle_deg` (or angle_rad) calculates the error metric based on angles.

closed 2019-07-16 by MichaelGrupp