MichaelGrupp/evo #189: how to recompile after editing the data

Roshnee, 2019-07-15:
I installed the editable version and edited the code in trajectory.py and i recompiled it to implement the changes. But the changes weren't applied. I would like to know how to do it in case i am doing something wrong. I know this is a basic question but I don't get it right. Please help

Roshnee, 2019-07-15:
I implemented the changes by changing the settings using `evo_config set` command
But it would be useful to know how to implement the changes made in code

MichaelGrupp, 2019-07-15:
If you installed with the `--editable` flag as described in the README, you should be able to use custom code changes directly without "recompiling". Maybe it works if you don't run pip again.

closed 2019-07-23 by MichaelGrupp