MichaelGrupp/evo #190: RPE calculation

ghost12111, 2019-07-15:
**Description:**
Even when only translational part is selected for evo_rpe tum, errors in rotational part are included in the result. If no --pose_relation argument is given, rotational errors also show up in a plot labeled translational part. This plot gives meters as its unit.

**Command:**
```
evo_rpe tum gt_test.txt res_test.txt -p --pose_relation trans_part
```
with gt_test.txt and res_test.txt being test files manually created to not have any translational error.

**Console output:**
```
RPE w.r.t. translation part (m)
for delta = 1 (frames) using consecutive pairs
(not aligned)

       max	2.000000
      mean	1.000000
    median	1.000000
       min	0.000000
      rmse	1.414214
       sse	8.000000
       std	1.000000

```

Please give also the following information:
* evo version number shown by `evo pkg --version`:  v1.5.1
* Python version shown by `evo pkg --pyversion`: 3.6.8
* operating system and version: Ubuntu 18.04
* did you change the source code? (yes / no):  no
* output of `evo_config show --brief --no_color`: 

```

{
    "console_logging_format": "%(message)s",
    "global_logfile_enabled": false,
    "plot_backend": "Qt5Agg",
    "plot_export_format": "pdf",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_multi_cmap": "none",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true
}
[gt_test.txt](https://github.com/MichaelGrupp/evo/files/3394231/gt_test.txt)
[res_test.txt](https://github.com/MichaelGrupp/evo/files/3394232/res_test.txt)

```

MichaelGrupp, 2019-07-15:
It's comparing local translations, not just position differences. 

If a frame is rotated or not _does_ make a difference for the translation.

In essence, your RPE scenario with the poses rotated by 90° looks like the image below. In the upper "trajectory", the translation from one pose to the next includes a negative component seen from the local frame while in the lower one it doesn't.

[image]

ghost12111, 2019-07-15:
Thank you for the quick response!
I falsely assumed that RPE would calculate the differences in the world reference frame much like APE.

closed 2019-07-15 by ghost12111