MichaelGrupp/evo #200: How can I draw a plot of specified parameter  of result.

TouchDeeper, 2019-08-17:
Hello,

I wanna draw a plot about rmse alone. Is there a convenient way to do this?

Thanks a lot!

MichaelGrupp, 2019-08-17:
I don't know if it's what you are searching for, but in the newest version I just added a way to configure which statistics are plotted in `evo_ape`, `evo_rpe` and `evo_res`:

```bash
evo_config set plot_statistics rmse
evo_config set plot_statistics rmse mean std
evo_config set plot_statistics none
# ...and so on, "evo_config reset" to get back to defaults.
```

That's at least how I find it useful :stuck_out_tongue_closed_eyes: 

If you need something more special I think the quickest way is to write a custom plot function yourself.

TouchDeeper, 2019-08-17:
Thank you for your reply @MichaelGrupp 

Could you please show me an example to get the RMSE results in the main_res.py? I just don't know where to get the data I want.

MichaelGrupp, 2019-08-17:
In `evo_res` the results are loaded into a [Pandas dataframe](https://pandas.pydata.org/pandas-docs/stable/reference/api/pandas.DataFrame.html): https://github.com/MichaelGrupp/evo/blob/e45574d3243c8d370fdeec4cf2b7066cee81b0ce/evo/main_res.py#L134

`df.loc["stats"]["rmse"]` should give you all the RMSE values of the results as a Pandas series.

In custom scripts you can also you simply use this:
```python
from evo.tools import file_interface

result = file_interface.load_res_file("your_file.zip")
rmse = result.stats["rmse"]
```

TouchDeeper, 2019-08-17:
Thank you a lot@MichaelGrupp

That is what I want. I have got the result that I want by a custom script.

closed 2019-08-17 by TouchDeeper