MichaelGrupp/evo #201: The legend of the plot will be out of order.

TouchDeeper, 2019-08-17:
Hello,

The legend of the plot will be out of order if the index is bigger than 9 as the picture below shows. Is there a way to put them in order?

[image]

Thanks a lot!

MichaelGrupp, 2019-08-17:
Fixed, the order will now be the same as you specify it on the command line.

closed 2019-08-17 by MichaelGrupp

TouchDeeper, 2019-08-17:
Thank you @MichaelGrupp

I use the way below to install the evo, could you please tell me how to modify the code in this installation way?

`pip install evo --upgrade --no-binary evo`

MichaelGrupp, 2019-08-17:
You can clone the repository and install it from source like described here: https://github.com/MichaelGrupp/evo#from-source
This allows you to change the code if you want.

If you just need this fix you can also wait until I push the next release and then upgrade with `pip install evo --upgrade --no-binary evo`. I just added a new one, so you should get it like that.

TouchDeeper, 2019-08-17:
Thank you @MichaelGrupp 

I will try the first way.

TouchDeeper, 2019-08-17:
Sorry @MichaelGrupp 

After I modified the code, Is it the right way to reinstall the evo for the code to work?

MichaelGrupp, 2019-08-17:
If you use `pip install --editable . --upgrade --no-binary evo` it means that you don't need to reinstall after you changed code. With `--editable`, pip links the installed files directly to your source folder.

TouchDeeper, 2019-08-17:
I get it. Thank you a lot @MichaelGrupp

flo981, 2020-02-16:
> Fixed, the order will now be the same as you specify it on the command line.

Seems not to be the case for bag files. The included topics appear in the legend not as specified.

flo981, 2020-02-16:
Same for TUM. After converting my bagfiles (as a work-around) to tum the order is still not like specified (see picture /sorry for the naming!)

[image]

evo pkg --version v1.5.2
evo pkg --pyversion 2.7.15+ (should change to python3?)
Ubuntu 16.04

MichaelGrupp, 2020-02-16:
> v1.5.2

This fix was included in 1.5.4

commit 0152a515 (2019-08-17): Use OrderedDict for trajectories in evo_traj (fixes #201). (#203)