MichaelGrupp/evo #210: problem about evaluating kitti data

CLM666, 2019-09-12:
**Description:**
I want to evaluate RMSE between kitti ground_truth file "00.txt"  and dso_result file "result_tum.txt". The former is kitti format , and I jave converted  it with "times.txt" to 00_truth_tum.txt by using your script. An confused problem is that it seems to be fine when I print the trajectory with  command evo_traj,   but the trajectory displayed is very confused when I used command evo_rpe. 

**Command:**
evo_traj tum 00_truth_tum.txt result_tum.txt --plot
evo_rpe tum 00_truth_tum.txt result_tum.txt --pose_relation angle_deg --delta 1 --delta_unit m --plot

**Console output:**
 evo_traj tum 00_truth_tum.txt result_tum.txt --plot
--------------------------------------------------------------------------------
name:	00_truth_tum
infos:	4541 poses, 3724.187m path length, 470.582s duration
--------------------------------------------------------------------------------
name:	result_tum
infos:	3635 poses, 4747.727m path length, 70.271s duration

evo_rpe tum 00_truth_tum.txt result_tum.txt --pose_relation angle_deg --delta 1 --delta_unit m --plot
RPE w.r.t. rotation angle in degrees (deg)
for delta = 1.0 (m) using consecutive pairs
(not aligned)

       max	143.706969
      mean	3.736700
    median	0.454811
       min	0.046648
      rmse	14.559756
       sse	99633.647806
       std	14.072084

**Additional files:**
Please attach all the files needed to reproduce the error.
[00_truth_tum.txt](https://github.com/MichaelGrupp/evo/files/3603448/00_truth_tum.txt)
[result_tum.txt](https://github.com/MichaelGrupp/evo/files/3603450/result_tum.txt)

Please give also the following information:
* evo version number shown by `evo pkg --version`:   v1.5.0
* Python version shown by `evo pkg --pyversion`: 3.6.8
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): ubuntu18.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
evo_config show --brief --no_color
{
    "console_logging_format": "%(message)s",
    "global_logfile_enabled": false,
    "plot_backend": "TkAgg",
    "plot_export_format": "pdf",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_multi_cmap": "none",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true
}

That's all of info about my issue, please help me if you are not busy.
Thak you very much!

CLM666, 2019-09-15:
I have solved it , thanks for your help very  much!

closed 2019-09-15 by CLM666

bryantaoli, 2020-05-02:
> I have solved it , thanks for your help very much!

how do you make the kitti ground truth to tum format:00_truth_tum？