MichaelGrupp/evo #214: Graphs don't match data

thohemp, 2019-09-17:
Hello,

I want to compare my TUM data test from orbslam2 with the groundtruth.
Using the avo_rpe I reveive a rmse of 0.02m.
But when I try to plot it both graphs are completely different:
[image]

MichaelGrupp, 2019-09-17:
Did you maybe run `evo_ape` with alignment and `evo_traj` without alignment?

thohemp, 2019-09-17:
Thanks for your answer. What do you mean with alignment? I use the output cameratrajectory from the fr1 room dataset and the corresponding groundtruth.

MichaelGrupp, 2019-09-18:
This here: https://github.com/MichaelGrupp/evo/wiki/evo_traj#geometric-alignment

If you're not sure please post your commands.

thohemp, 2019-09-18:
Hm, i used following command for plotting:

`evo_traj tum CameraTrajectory.txt groundtruth.txt -p`

which results in the plot above.

For rpe I used:
`evo_rpe tum CameraTrajectory.txt groundtruth.txt`

with following output:

> RPE w.r.t. translation part (m)
for delta = 1 (frames) using consecutive pairs
(not aligned)

       max      0.112366
      mean      0.008361
    median      0.006333
       min      0.000243
      rmse      0.011756
       sse      0.186722
       std      0.008264

thohemp, 2019-09-18:
Okay, according to your link, I used:
`evo_traj tum CameraTrajectory.txt --ref groundtruth.txt -p --align_origin`

which gives m
[image]
e a much better result plot:

I am doing this right?

MichaelGrupp, 2019-09-18:
Ah, I misread your text. You are using `evo_rpe`, not `evo_ape`.

The relative pose error doesn't change if the trajectories are aligned or not because it's comparing _relative_ poses. So your results are expected.

closed 2019-09-26 by MichaelGrupp