MichaelGrupp/evo #215: when I run "evo_traj tum " some errors occur

heartLH, 2019-09-27:
**Description:**
when I run the command:

**Command:**
```
# evo_traj tum pose_output2.txt -p --plot_mode=xyz
```

**Console output:**
```
[ERROR] TUM trajectory files must have 8 entries per row and no trailing delimiter at the end of the rows (space)

```

**Additional files:**
but I don't think my tumdataset is error in format.
This error occurs even if it contains only one row of data.
1403638522.827830 0.020021 0.005823 0.182206 -0.055571 0.818886 0.023182 0.570789

Here are my information:
* evo version 1.5.5
* Python version shown by `3.6.8`
* operating system and version Ubuntu 18.04 
* did you change the source code?  no
* output of `evo_config show --brief --no_color`: 
"console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_backend": "Qt5Agg",
    "plot_export_format": "pdf",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_multi_cmap": "none",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true

```
who can help me?

MichaelGrupp, 2019-09-27:
The error already tells that the line should not have an extra space at the end ("`no trailing delimiter at the end of the rows (space)`"), so it looks like you need to remove it: https://github.com/MichaelGrupp/evo/issues/70#issuecomment-402031390

closed 2019-11-18 by MichaelGrupp

yinhaosen, 2023-04-19:
How to check trailing delimiter at the end of rows (space)? I encountered the same problem and used `vim` to check the file but there was no space at the end of each row at all.

MichaelGrupp, 2023-04-19:
@yinhaosen Can you upload the file here?

yinhaosen, 2023-04-19:
> @yinhaosen Can you upload the file here?

Sure.
[pose_modified.csv](https://github.com/MichaelGrupp/evo/files/11271628/pose_modified.csv)

yinhaosen, 2023-04-20:
> Your file has ~70000 lines with ` 0 0 ` at the end.

Oh shit! My bad, I didn't even check the end. You are right. Thanks for pointing it out. I guess there is something wrong with data post-processing.