MichaelGrupp/evo #217: Cannot find evo pkg

artemiialessandrini, 2019-10-16:
**Description: After installation from source all seem to be succesfully done, but command evo_traj not found**

**Command:**
```
~/evo/test/data$ evo_traj tum result.tum -p --plot_mode=xy
```

**Console output:**
```

evo_traj: command not found

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

Ubuntu 18.04, 5.0.0-29-generic
Command 'evo' not found, did you mean...
evo_config: command not found

```

artemiialessandrini, 2019-10-16:
`pip show --files evo` output:
```
Name: evo
Version: 1.5.5
Summary: Python package for the evaluation of odometry and SLAM
Home-page: https://github.com/MichaelGrupp/evo
Author: Michael Grupp
Author-email: michael.grupp@tum.de
License: GPLv3
Location: /home/artemii/evo
Requires: numpy, matplotlib, scipy, pandas, seaborn, natsort, argcomplete, colorama, pygments
Files:
Cannot locate installed-files.txt
```
Also, kindly find
[`pip3 install --editable . --upgrade --no-binary evo output`](https://gist.github.com/artemiialessandrini/e21b0fb616fb2f144dacee1e41745705)

MichaelGrupp, 2019-10-17:
Can you try to run the pip installation again with `--verbose`? It should print where files are being installed to.

`pip3 install --editable . --upgrade --no-binary evo output --verbose`

I'm also not sure if there are different results with pip / pip3

artemiialessandrini, 2019-10-17:
Artemii Garibov,Robotics Software Engineer, MechatronicsHaidian, Beijing, China

artemiialessandrini, 2019-11-15:
Somehow issue got solved, all works good!
I reinstalled Eigen to the last version recently, if it could be somehow related..

AgarwalVibha, 2019-11-20:
I am having this same issue! 
I used: 
pip install --editable . --upgrade --no-binary evo 
and the error was: 
evo_traj: command not found

artemiialessandrini, 2019-11-21:
@AgarwalVibha , good, then let's wait anyone to try again if it fixes the issue, so it can be closed.

liuzhenboo, 2020-02-02:
> I am having this same issue!
> I used:
> pip install --editable . --upgrade --no-binary evo
> and the error was:
> evo_traj: command not found

when I restarted my computer, it works!
`lzb@lzb:~$ evo --version`
`Initialized new /home/lzb/.evo/settings.json`
`usage: evo [-h] {pkg,cat_log} ...`
`evo: error: too few arguments``

LarryDong, 2020-02-09:
> when I restarted my computer, it works!
> `lzb@lzb:~$ evo --version`
> `Initialized new /home/lzb/.evo/settings.json`
> `usage: evo [-h] {pkg,cat_log} ...`
> `evo: error: too few arguments``

For me, reboot the system and evo commands can be found.

MichaelGrupp, 2020-03-30:
If someone encounters this again, please attach the output of pip with the `--verbose` flag activated.

Other than that, it's probably caused by a corrupt Python environment.

closed 2020-03-30 by MichaelGrupp

abhy1012, 2020-04-15:
Hi having the same error.  when I do pip install it says evo is installed but when I call executable it doesn't work.

this is my verbose output:

DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Defaulting to user installation because normal site-packages is not writeable
Created temporary directory: /private/var/folders/nj/rbpthdt53j504g_97jm42l7h0000gn/T/pip-ephem-wheel-cache-wbOorZ
Created temporary directory: /private/var/folders/nj/rbpthdt53j504g_97jm42l7h0000gn/T/pip-req-tracker-3lo351
Initialized build tracking at /private/var/folders/nj/rbpthdt53j504g_97jm42l7h0000gn/T/pip-req-tracker-3lo351
Created build tracker: /private/var/folders/nj/rbpthdt53j504g_97jm42l7h0000gn/T/pip-req-tracker-3lo351
Entered build tracker: /private/var/folders/nj/rbpthdt53j504g_97jm42l7h0000gn/T/pip-req-tracker-3lo351
Created temporary directory: /private/var/folders/nj/rbpthdt53j504g_97jm42l7h0000gn/T/pip-install-VRJN_F
Obtaining file:///Users/amyt/Downloads/evo-master
  Added file:///Users/amyt/Downloads/evo-master to build tracker '/private/var/folders/nj/rbpthdt53j504g_97jm42l7h0000gn/T/pip-req-tracker-3lo351'
    Running setup.py (path:/Users/amyt/Downloads/evo-master/setup.py) egg_info for package from file:///Users/amyt/Downloads/evo-master
    Running command python setup.py egg_info
    /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
      warnings.warn(msg)
    /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/setuptools/dist.py:285: UserWarning: Normalizing 'v1.7.2
    ' to '1.7.2'
      normalized_version,
    fastentrypoints
    running egg_info
    writing requirements to evo.egg-info/requires.txt
    writing evo.egg-info/PKG-INFO
    writing top-level names to evo.egg-info/top_level.txt
    writing dependency_links to evo.egg-info/dependency_links.txt
    writing entry points to evo.egg-info/entry_points.txt
    reading manifest file 'evo.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file 'evo.egg-info/SOURCES.txt'
  Source in /Users/amyt/Downloads/evo-master has version 1.7.2, which satisfies requirement evo==1.7.2 from file:///Users/amyt/Downloads/evo-master
  Removed evo==1.7.2 from file:///Users/amyt/Downloads/evo-master from build tracker '/private/var/folders/nj/rbpthdt53j504g_97jm42l7h0000gn/T/pip-req-tracker-3lo351'
Requirement already satisfied, skipping upgrade: numpy in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from evo==1.7.2) (1.8.0rc1)
Requirement already satisfied, skipping upgrade: matplotlib in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from evo==1.7.2) (1.3.1)
Requirement already satisfied, skipping upgrade: scipy>=1.2 in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from evo==1.7.2) (1.2.3)
Requirement already satisfied, skipping upgrade: pandas in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from evo==1.7.2) (0.24.2)
Requirement already satisfied, skipping upgrade: seaborn>=0.9 in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from evo==1.7.2) (0.9.1)
Requirement already satisfied, skipping upgrade: natsort in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from evo==1.7.2) (6.2.1)
Requirement already satisfied, skipping upgrade: argcomplete in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from evo==1.7.2) (1.11.1)
Requirement already satisfied, skipping upgrade: colorama>=0.3 in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from evo==1.7.2) (0.4.3)
Requirement already satisfied, skipping upgrade: pygments in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from evo==1.7.2) (2.5.2)
Requirement already satisfied, skipping upgrade: pyyaml in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from evo==1.7.2) (5.3.1)
Requirement already satisfied, skipping upgrade: pillow in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from evo==1.7.2) (6.2.2)
Requirement already satisfied, skipping upgrade: enum34 in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from evo==1.7.2) (1.1.10)
Requirement already satisfied, skipping upgrade: python-dateutil in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from matplotlib->evo==1.7.2) (1.5)
Requirement already satisfied, skipping upgrade: tornado in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from matplotlib->evo==1.7.2) (5.1.1)
Requirement already satisfied, skipping upgrade: pyparsing>=1.5.6 in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from matplotlib->evo==1.7.2) (2.0.1)
Requirement already satisfied, skipping upgrade: nose in /Users/amyt/Library/Python/2.7/lib/python/site-packages (from matplotlib->evo==1.7.2) (1.3.7)
Requirement already satisfied, skipping upgrade: pytz>=2011k in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from pandas->evo==1.7.2) (2013.7)
[... cut ...]
    Installing evo_rpe script to /Users/amyt/Library/Python/2.7/bin
    Installing evo_config script to /Users/amyt/Library/Python/2.7/bin
    Installing evo_ipython script to /Users/amyt/Library/Python/2.7/bin
    Installing evo_res script to /Users/amyt/Library/Python/2.7/bin
    Installing evo_fig script to /Users/amyt/Library/Python/2.7/bin

    Installed /Users/amyt/Downloads/evo-master
Successfully installed evo
Cleaning up...
Removed build tracker: '/private/var/folders/nj/rbpthdt53j504g_97jm42l7h0000gn/T/pip-req-tracker-3lo351'

MichaelGrupp, 2020-04-15:
@abhy1012 pip did an installation to a user directory in your case, e.g. "Installing evo_traj script to /Users/amyt/Library/Python/2.7/bin"

If you add that path to your `PATH` environment variable it should work. See here: https://stackoverflow.com/a/43368894

wxchencn, 2020-05-03:
If you use ubuntu on WSL(Windows Subsystem for Linux), you will find evo installed in `~/.local/bin`. 
add it to `PATH` !

srinivasrama, 2022-04-12:
Please reboot the system

arjunskumar, 2022-08-05:
Thanks @srinivasrama ,it worked

foifoipipi, 2023-01-12:
Hello.
I rebooted my system and could not find the evo command .... I have tried all the above mentioned methods but it doesn't work. Can someone please provide me with a solution?