MichaelGrupp/evo #218: File seems to be good, but still "TUM trajectory files must have 8 entries per row and no trailing delimiter at the end of the rows (space)"

artemiialessandrini, 2019-10-16:
**Description:**

**Command:**
```
# replace this line with the command(s) you used
```

**Console output:**
```

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```

artemiialessandrini, 2019-10-16:
Btw, I installed evo in another Ubuntu 16.04, so the previous question is still relevant.

artemiialessandrini, 2019-10-16:
Stupid mistake, I don't have **tz**

artemiialessandrini, 2019-10-16:
lovely
[image]

closed 2019-10-16 by artemiialessandrini

genius-hl, 2021-04-23:
hello,i have the `Error:KITTI pose files must have 12 entries per row and no trailing delimiter at the end of the rows (space)` when i run the command `evo_traj kitti KeyFrameTrajectory.txt --ref=KITTI_00_gt.txt -p --plot_mode=xz`.And i try to solve it by `cut --delimiter " " --fields 1-8 KeyFrameTrajectory.txt > KeyFrameTrajectory.txt `,then there is nothing in the file.How can i sovle the above problem?Thank you for your help in advance!

MichaelGrupp, 2021-04-23:
@genius-hl you are trying to do something different with KITTI - it doesn't make sense to blindly run the `cut ...` command that was mentioned here. You should rather check if your file contains the correct data in the right format. This might help: https://github.com/MichaelGrupp/evo/wiki/Formats#kitti---kitti-dataset-pose-format

genius-hl, 2021-04-23:
> @genius-hl you are trying to do something different with KITTI - it doesn't make sense to blindly run the `cut ...` command that was mentioned here. You should rather check if your file contains the [...]

Thank you for your reply! I solve the problem!!

wcm-123, 2021-12-05:
> 

> lovely [image]

Hello, can you send me a copy of the data set corresponding to this graph, thank you!

artemiialessandrini, 2021-12-05:
Hi, @wcm-123, [ekftagscarto0.txt](https://github.com/MichaelGrupp/evo/files/7656320/ekftagscarto0.txt) can you access this one?

wcm-123, 2021-12-05:
> Hi, @wcm-123, [ekftagscarto0.txt](https://github.com/MichaelGrupp/evo/files/7656320/ekftagscarto0.txt) can you access this one?
thank you.I can download it, but what I mean is that this picture of you is a data set package similar to euroc made by your own experiment, I want this data set package to test my VIO algorithm