MichaelGrupp/evo #219: How to change linetype of estimated data

CLM666, 2019-10-16:
Hi, I like evo very much, and I found when I run "evo_traj"  to print multiple estimated file, they have same line type. the line type of reference can be changed  by "evo_config", but  I don't find how to change the line type of estimated file. Can you help me? Thank you very much!

MichaelGrupp, 2019-10-17:
The line type of normal trajectories is currently not available as a quick setting in `evo_config`. But I will try to add this soon, shouldn't be much work.

> when I run "evo_traj" to print multiple estimated file, they have same line type

Do you want each one to have a different line type? I don't think I will add a feature for this, but you can change the type (currently `'-'` for all trajectories) in the code if you want to customize it: https://github.com/MichaelGrupp/evo/blob/d004a179a5f95a1f003efbe79993ba69a5747341/evo/main_traj.py#L414

MichaelGrupp, 2019-10-18:
I added the `plot_trajectory_linestyle` parameter in the newest version for completeness. But as I said, it's maybe not what you need if you want all trajectories to have a different linestyle.

CLM666, 2019-10-21:
Ok, I get that. Thanks for your quick reply and excellent work!

closed 2019-11-18 by MichaelGrupp