MichaelGrupp/evo #221: Using Evo to evaluate slam from bag

msadowski, 2019-11-21:
Hi!

I have just been trying to check the quality of my SLAM from a rosbag. I think the main issue is that most ROS slam packages provide map->odom tf transform in the SLAM process. Looking at the supported messages I don't think this can be supported out of the box. Do you have a workflow for this already? With my current bag files only the odometry/filtered topics are found, which understandably don't carry any useful information about the quality of SLAM.

It looks to me like this script: https://github.com/MichaelGrupp/evo/pull/44/files will probably solve my issue but I'd rather check if you have any other experiences or can suggest an alternative approach.

Thanks a lot for all the work on this tool!

MichaelGrupp, 2019-11-21:
I guess mean TF topics of the type `tf/tfMessage` / `tf2_msgs/TFMessage`? They are indeed not supported out of the box because in order to select specific trajectories one would need to specify the parent and child frame IDs of each trajectory, which can get tedious.

evo does support `geometry_msgs/TransformStamped` though. Since a `TFMessage` is nothing but a list of `geometry_msgs/TransformStamped` messages, it should be straightforward to write a tool that converts a bagfile with a TF topic into a bag with a bunch of `geometry_msgs/TransformStamped` topics (one for each transformation e.g. "/map_to_odom").

> It looks to me like this script: https://github.com/MichaelGrupp/evo/pull/44/files will probably solve my issue

You can use that for recording a TF transformation's trajectory, yes.

msadowski, 2019-11-21:
Yes, I meant the tfMessages! 

The python script I linked worked really well with my bag - just in my case I had to do one additional step as I wanted to compare the data from two bag files. I'll describe the workflow fully on my blog and link it here in case someone wants to reproduce it.

closed 2019-11-21 by msadowski

MichaelGrupp, 2019-11-21:
Awesome, feel free to also send a PR to add your article to the "Supported trajectory formats:" part of the README.

BTW, I would recommend to use a link to the latest master version instead of the one you used:
https://github.com/MichaelGrupp/evo/blob/master/contrib/record_tf_as_posestamped_bag.py

msadowski, 2019-11-26:
@MichaelGrupp I described the provess on my blog in this post: https://msadowski.github.io/Comparing-SLAM-with-ROS-evo/ . Since the process is quite convoluted at the moment I'm not sure it's worth mentioning it in the readme. 

I will check how my future evaluations look like - I have a feeling it might be worth adding a support for handling the tf messages directly - this would eliminate one of the steps I had shown in the post.

MichaelGrupp, 2019-11-26:
Ok. Supporting a /tf topic directly would be nice, but as I said I'm not sure how we could include this into the command line interface nicely... 

Anyway, I made a follow up feature request: https://github.com/MichaelGrupp/evo/issues/222

MichaelGrupp, 2020-06-05:
@msadowski TF transformations from bags are now directly supported since v1.9: https://github.com/MichaelGrupp/evo/commit/63cdfc8cbbb0a388272ce48bc2fcba6af48e3adb

You need to pass a special ID for the topic/transform combination you want to load, see here:
https://github.com/MichaelGrupp/evo/wiki/Formats#bag---ros-bagfile

Bag merging is not there though, but I guess there are enough other tools out there to do that job.

msadowski, 2020-06-09:
That's amazing! Many thanks!!!!

I think it's good you are not merging bags, there are other tools for that :). Next time I have access to a robot with decent odometry sources I'll try it out!

MichaelGrupp, 2020-06-10:
Cool, looking forward to the results :smile: