MichaelGrupp/evo #223: cannot find evo_ape

liubamboo, 2019-11-29:
**Description:**
I use command "pip install evo --upgrade --no-binary evo" to install evo.
then, I run evo_ape --help, but it cannot find the "evo_ape"

**Command:**
```
evo_ape --help
```

**Console output:**
```
evo_ape：未找到命令(cannot find the command)
```

Please give also the following information:
* evo version number shown by `evo pkg --version`: evo-1.5.7(actually the command "evo pkg --version" doesn't work. I find the version of evo when install evo)
* Python version shown by `evo pkg --pyversion`: it show "Command 'evo' not found",when I run the command "python" it show Python 2.7.15+ (default, Oct  7 2019, 17:39:04)
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): ubuntu 18
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: evo_config：未找到命令(cannot find the command)

liubamboo, 2019-11-29:
Luckyly, when I restart ubuntu, I work!!!

MichaelGrupp, 2019-11-29:
Looks similar to #217, unfortunately I can't say more at the moment. Could be a problem with pip.

But good that a restart solved it for you!

closed 2019-11-29 by MichaelGrupp