MichaelGrupp/evo #224: Absilute pose error looks wrongly calculated

artemiialessandrini, 2019-12-06:
Hi
I'm comparing trajectories using:
```evo_ape tum robot_pose_origin.tum 2019-12-06-13-41-13.txt --plot --plot_mode=xy```,

But seems that absolute trajectory result looks off from the actual one. I'm using trajectory, extracted from cartographer and trajectory from Ground Truth.

[image]
[image]
Any ideas, maybe I'm missing something?

There is a problem with stamps amount, it's **quite** different numbers. ( only 219 from cartographer extraction and 7386 from GT) 
Could it be the reason of this issue?

Attaching trajectory tum files and command line output:
[2019-12-06-13-41-13.txt](https://github.com/MichaelGrupp/evo/files/3931182/2019-12-06-13-41-13.txt)
[robot_pose_origin.txt](https://github.com/MichaelGrupp/evo/files/3931183/robot_pose_origin.txt)

```--------------------------------------------------------------------------------
Loaded 7386 stamps and poses from: robot_pose_origin.tum
Loaded 221 stamps and poses from: 2019-12-06-13-41-13.txt
Synchronizing trajectories...
Found 219 of max. 7386 possible matching timestamps between...
	robot_pose_origin.tum
and:	2019-12-06-13-41-13.txt
..with max. time diff.: 0.01 (s) and time offset: 0.0 (s).
--------------------------------------------------------------------------------
Compared 219 absolute pose pairs.
Calculating APE for translation part pose relation...
--------------------------------------------------------------------------------
APE w.r.t. translation part (m)
(not aligned)

       max	0.205512
      mean	0.201863
    median	0.202285
       min	0.200070
      rmse	0.201867
       sse	8.924307
       std	0.001314

--------------------------------------------------------------------------------
Plotting results... 
```

MichaelGrupp, 2019-12-06:
Without knowing your precise setup, there might be an offset between your SLAM and groundtruth coordinate system, e.g. caused by calibration. You can try to use [alignment](https://github.com/MichaelGrupp/evo/wiki/Metrics#alignment) to minimize such effects when calculating the metrics.

artemiialessandrini, 2019-12-09:
Thank you for your response, @MichaelGrupp , please correct me if I'm wrong for clarification

I assume, plotting is only based on trajectory files (e.g. tum, KITTI etc.) data provided
Or does that mean, that evo tool takes into account coordinate frames somehow?

I'd be happy to get trajectories in one coordinate frame without alignments, so having a proper comparing.

artemiialessandrini, 2019-12-09:
Or do you mean time offset?

MichaelGrupp, 2019-12-09:
> I'd be happy to get trajectories in one coordinate frame without alignments, so having a proper comparing.

Then the input data must be already in the same coordinate frame.

> Or does that mean, that evo tool takes into account coordinate frames somehow?

No, it doesn't. So, depending on the source data and what you want to achieve, either the data must be already in the correct frame or can be aligned.

> Or do you mean time offset?

No, I mean geometric offset. But as I said, this completely depends on your personal groundtruth/SLAM setup. Just as additional info: time offsets can have an effect on alignment though (see [here](https://github.com/MichaelGrupp/evo/issues/117)), but I think this is not relevant to your question at the moment.

closed 2019-12-22 by MichaelGrupp