MichaelGrupp/evo #226: kitty pose file

Shivani1796, 2019-12-19:
I am working on a project in which i need Monocular Visual Odometry by my own dataset but i am confused that how to create a pose file like kitty dataset 00.txt , can you help me in this? 
Thanks ..

MichaelGrupp, 2019-12-19:
I can't help you with building your visual odometry code, but you can find an overview of the contents of trajectory file formats here: https://github.com/MichaelGrupp/evo/wiki/Formats

Shivani1796, 2019-12-19:
@MichaelGrupp  Thanks for responding, but already visited the above site and need to build pose file for my dataset . Do you know how to make pose file ?

MichaelGrupp, 2019-12-28:
It's just a text file with the pose data in each row.

closed 2019-12-28 by MichaelGrupp

Shivani1796, 2019-12-30:
@MichaelGrupp  But how to obtain that text file ?

SaiArunesh, 2021-04-16:
It's a text file that should have the Ground truth poses of your dataset for comparison