MichaelGrupp/evo #238: How to save evo_traj data in latex table format?

ricber, 2020-01-30:


MichaelGrupp, 2020-01-30:
Currently there's no table export feature in `evo_traj`. Only `evo_res` has this (see the example in the README) and you can configure the `table_export_format` to "latex" in the settings.

> I was expecting to see a --save_trajectory SAVE_TRAJECTORY option.

This sounds like you want to save the trajectory in Latex? :thinking: Or did you mean the trajectory infos/statistics that are printed? (like here https://github.com/MichaelGrupp/evo/wiki/evo_traj#check-trajectory-data)

ricber, 2020-01-31:
> 
> 
> Currently there's no table export feature in `evo_traj`. Only `evo_res` has this (see the example in the README) and you can configure the `table_export_format` to "latex" in the settings.
> 
> > I was expecting to see a --save_trajectory SAVE_TRAJECTORY option.
> 
> This sounds like you want to save the trajectory in Latex? 🤔 Or did you mean the trajectory infos/statistics that are printed? (like here https://github.com/MichaelGrupp/evo/wiki/evo_traj#check-trajectory-data)

Sorry, I wrote wrong. I meant the `--save_table SAVE_TABLE` option.

MichaelGrupp, 2020-06-28:
`--save_table` was added in the latest version. Latex can be used by setting `evo_config set table_export_format latex`.

closed 2020-06-28 by MichaelGrupp