MichaelGrupp/evo #242: Error when using evo_res results/*.zip -p --save_table results/table.csv

lyrem0007a, 2020-02-07:
**Description:**
Hi,

I got an error when I use the evo_res for evaluation if several zip files. I have windows 10 with python 3.7.4. I got following traceback:

**Command:**
```
evo_res results/*.zip -p --save_table results/table.csv
```

**Console output:**
```

Traceback (most recent call last):
  File "C:\Users\test\.conda\envs\venv\lib\site-packages\evo\entry_points.py", line 92, in launch
    main_module.run(args)
  File "C:\Users\test\.conda\envs\venv\lib\site-packages\evo\main_res.py", line 135, in run
    args.merge)
  File "C:\Users\test\.conda\envs\venv\lib\site-packages\evo\main_res.py", line 108, in load_results_as_dataframe
    result = file_interface.load_res_file(result_file)
  File "C:\Users\test\.conda\envs\venv\lib\site-packages\evo\tools\file_interface.py", line 353, in load_res_file
    with zipfile.ZipFile(zip_path, mode='r') as archive:
  File "C:\Users\test\.conda\envs\venv\lib\zipfile.py", line 1207, in __init__
    self.fp = io.open(file, filemode)
OSError: [Errno 22] Invalid argument: 'results/*.zip'

[ERROR] evo module evo.main_res crashed - no logfile written (disabled)

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`:  v1.7
* Python version shown by `evo pkg --pyversion`: 3.7.4
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Windows 10
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_export_format": "pdf",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_multi_cmap": "none",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_unknown_cell_value": 205,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true
}
```

MichaelGrupp, 2020-02-07:
I guess it's because you run on Windows and use a terminal that doesn't support the `*` wildcard notation. Either you need to write the full file names instead of *.zip or use sth. like PowerShell or Bash on Windows.

lyrem0007a, 2020-02-07:
Thanks a lot!!!

Beste Grüße aus München

closed 2020-02-07 by lyrem0007a

MichaelGrupp, 2020-02-07:
Gruß zurück :grin:

jingshaojing, 2021-01-21:
I have the same question,but  I use linux

Tejal-19, 2023-07-31:
Same. Tried writing the full name as well. Issue still persists

MichaelGrupp, 2023-07-31:
Please provide a bit more details. The original issue here was about using wildcard `*` notation on Windows and was resolved by not doing this, so not sure what's the relation.

Tejal-19, 2023-08-01:
[ERROR] Unhandled error in evo.main_res
Traceback (most recent call last):
  File "/home/miko/.local/lib/python3.8/site-packages/evo/entry_points.py", line 99, in launch
    main_module.run(args)
  File "/home/miko/.local/lib/python3.8/site-packages/evo/main_res.py", line 267, in run
    dist_grid.map(sns.distplot, metric_label)  # fits=stats.gamma
  File "/home/miko/.local/lib/python3.8/site-packages/seaborn/axisgrid.py", line 752, in map
    self._facet_plot(func, ax, plot_args, kwargs)
  File "/home/miko/.local/lib/python3.8/site-packages/seaborn/axisgrid.py", line 848, in _facet_plot
    func(*plot_args, **plot_kwargs)
  File "/home/miko/.local/lib/python3.8/site-packages/seaborn/distributions.py", line 2511, in distplot
    kdeplot(**{axis: a}, ax=ax, color=kde_color, **kde_kws)
  File "/home/miko/.local/lib/python3.8/site-packages/seaborn/distributions.py", line 1717, in kdeplot
    p.plot_univariate_density(
  File "/home/miko/.local/lib/python3.8/site-packages/seaborn/distributions.py", line 999, in plot_univariate_density
    artist, = ax.plot(support, density, **artist_kws)
  File "/usr/lib/python3/dist-packages/matplotlib/axes/_axes.py", line 1665, in plot
    lines = [*self._get_lines(*args, data=data, **kwargs)]
  File "/usr/lib/python3/dist-packages/matplotlib/axes/_base.py", line 225, in __call__
    yield from self._plot_args(this, kwargs)
  File "/usr/lib/python3/dist-packages/matplotlib/axes/_base.py", line 386, in _plot_args
    x = _check_1d(tup[0])
  File "/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py", line 1402, in _check_1d
    ndim = x[:, None].ndim
  File "/home/miko/.local/lib/python3.8/site-packages/pandas/core/indexes/base.py", line 5199, in __getitem__
    disallow_ndim_indexing(result)
  File "/home/miko/.local/lib/python3.8/site-packages/pandas/core/indexers/utils.py", line 343, in disallow_ndim_indexing
    raise ValueError(
ValueError: Multi-dimensional indexing (e.g. `obj[:, None]`) is no longer supported. Convert to a numpy array before indexing instead.

[ERROR] evo module evo.main_res crashed - no logfile written (disabled)

evo pkg --version 
v1.23.1
evo pkg --pyversion 
3.8.10
evo_config show --brief --no_color
{
    "console_logging_format": "%(message)s",
    "euler_angle_sequence": "sxyz",
    "global_logfile_enabled": false,
    "plot_axis_marker_scale": 0.0,
    "plot_backend": "Qt5Agg",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_mode_default": "xyz",
    "plot_multi_cmap": "none",
    "plot_pose_correspondences": false,
    "plot_pose_correspondences_linestyle": "dotted",
    "plot_reference_alpha": 0.5,
    "plot_reference_axis_marker_scale": 0.0,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_show_axis": true,
    "plot_show_legend": true,
    "plot_split": false,
    "plot_start_end_markers": false,
    "plot_statistics": [
        "rmse",
        "median",
        "mean",
        "std",
        "min",
        "max"
    ],
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_trajectory_linestyle": "-",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "ros_map_alpha_value": 1.0,
    "ros_map_cmap": "Greys_r",
    "ros_map_enable_masking": true,
    "ros_map_unknown_cell_value": 205,
    "ros_map_viewport": "keep_unchanged",
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true,
    "tf_cache_lookup_frequency": 10,
    "tf_cache_max_time": 10000.0
}

Tejal-19, 2023-08-01:
I get the following error when I try the following command 
evo_res results/result_1.zip results/result_2.zip -p --save_table results/table_3.csv