MichaelGrupp/evo #243: evo_traj: Multiple rosbags / no matching timestamps

flo981, 2020-02-07:
Hey, 
as far as I know, there is no way to load more than one bag file to evo_traj.
I have two bag files with pose and groundtruth. Each bag file has different time stamps since they are recorded after each other. 

```
evo_traj bag test.bag /stereo_pose_orb_1 /stereo_pose_orb_2 --ref /groundtruth -p -as
[ERROR] found no matching timestamps between reference and /stereo_pose_orb_1 with max. time diff 0.01 (s) and time offset 0.0 (s)
```
groundtruth is recorded together with stereo_pose_orb_2 (matching time stampes) while stereo_pose_orb_1 was recorded before.

I played both bag files simultaneously  and re-recorded everything in one bagfile, but the _no matching timestamps_ issue is still there.

Is there any way to plot both poses and the groundtruth in one plot? (if possible without converting it to kitti/tum)
Thanks in advance!

MichaelGrupp, 2020-02-07:
> Each bag file has different time stamps since they are recorded after each other.

If this is the case, you either need to account for the offset or better record both with the same clock (related issue: https://github.com/MichaelGrupp/evo/issues/16).

MichaelGrupp, 2020-02-07:
> Is there any way to plot both poses and the groundtruth in one plot? (if possible without converting it to kitti/tum)

If the timestamps are different, only without the alignment options. 

Multiple bagfiles are not supported in the command line apps at the moment. This _could_ be added as a feature theoretically, the underlying function just expects a `rosbag.Bag()` handle and we could have multiple of them.

flo981, 2020-02-07:
Okay thanks!
Wasn't there something like `--use_rel_time` in case timestamps are not matching? Can remember using something like this but can't recap it or find it anymore

flo981, 2020-02-11:
Is there a way to export the resulting trajectory of

`evo_traj bag file.bag /topic1 --ref /topic2 -as` ?

to a .csv file? Thus I can use the aligned and scaled results from different bag files and plot them e.g. in matlab.

MichaelGrupp, 2020-02-11:
You can save trajectories in different formats: https://github.com/MichaelGrupp/evo/wiki/Formats#saving--exporting-to-other-formats
"TUM" and "KITTI" are essentially just csv-like files, as described on the same page.

Plotting everything together is just a matter of calling `evo_traj tum <your files> -p`.

> Wasn't there something like --use_rel_time in case timestamps are not matching? Can remember using something like this but can't recap it or find it anymore

It exists only in `evo_res` at the moment and means that for each set of timestamps the first timestamp is subtracted from all timestamps, so they all start at 0. It wouldn't be hard to add sth like this also to evo_traj. Although the cleaner solution is to use the same clock for all trajectories, as mentioned before.

closed 2020-02-16 by flo981