MichaelGrupp/evo #247: APE and ATE

EnriqueSolarte, 2020-02-13:
Dear all, 
 I want to ask if the APE (absolute pose error) metric means the same as ATE (absolute trajectory error)?. The second one was the metric introduced by TUM in their dataset. 
In advance, thank you.

MichaelGrupp, 2020-02-14:
Using the "APE" of evo with alignment and translational pose relation is equivalent to the "ATE" described in the TUM RGB-D paper:

```
evo_ape tum <reference> <estimated> --align --pose_relation trans_part
```
There's also a direct comparison in this Jupyter notebook: https://github.com/MichaelGrupp/evo/blob/master/notebooks/metrics.py_API_Documentation.ipynb

closed 2020-02-28 by MichaelGrupp

saqib1707, 2020-07-02:
@MichaelGrupp 

While trying to understand the relation between ATE defined in [TUM RGB-D paper](https://ieeexplore.ieee.org/document/6385773) and EVO APE with translational pose relation, I came across this issue: 

In [this part of the code](https://github.com/MichaelGrupp/evo/blob/master/evo/core/trajectory.py#L356-L361), the conversion from xyz-quat to SE(3) camera pose, the xyz component is included in the 4x4 transformation matrix as it is whereas in my opinion, the translational part of any 4x4 SE(3) pose matrix equals -R (SO(3) rotation matrix) times the xyz coordinate of camera center. Even in the [reverse conversion](https://github.com/MichaelGrupp/evo/blob/master/evo/core/trajectory.py#L364-L367), the translational part of SE(3) pose is directly taken as xyz coordinate without left-multiplication with $-R^{T}$.

Even [here](https://github.com/MichaelGrupp/evo/blob/master/evo/core/metrics.py#L366-L368), the xyz coordinates of the estimated and reference camera center are directly subtracted. Whereas, according to this [paper](https://ieeexplore.ieee.org/document/6385773), it should be the translational component of the relative pose $E_i = Q_i^{-1} P_i$. 

I am not sure whether there is an actual error or I am missing something here. Can you please clarify ? Thank you very much.

MichaelGrupp, 2020-07-08:
>  the translational part of any 4x4 SE(3) pose matrix equals -R (SO(3) rotation matrix) times the xyz coordinate of camera center

This is wrong. Also, there's no notion of a camera center here, we're dealing generic poses expressed by SE(3) matrices. They could be a camera pose, but this doesn't matter in this context.

> Even here, the xyz coordinates of the estimated and reference camera center are directly subtracted. Whereas, according to this paper, it should be the translational component of the relative pose $E_i = Q_i^{-1} P_i$

The length of the distance vector is the same. And this is what matters here in the code a few lines later.