MichaelGrupp/evo #248: Invalid quaternion

IoannisKaragiannis, 2020-02-17:
**Description:**
Hi,

I have downloaded the MH_01_easy (ASL format) from the EuRoC dataset and I want to convert it from csv to tum format and check that everything is fine. For some reason I am getting quaternions in an invalid format. Any idea why? According to your instructions I have done the following

**Command:**
```
evo_traj euroc ~/Downloads/MH_01_easy/mav0/state_groundtruth_estimate0/data.csv --save_as_tum
```

**Console output:**
```

name:	data
infos:
	duration (s)	181.904999971
	nr. of poses	36382
	path length (m)	80.6262118473
	pos_end (m)	[ 4.572531 -1.942023  0.61962 ]
	pos_start (m)	[ 4.688319 -1.786938  0.783338]
	t_end (s)	1403636762.74
	t_start (s)	1403636580.84
checks:
	SE(3) conform	yes
	array shapes	ok
	nr. of stamps	ok
	quaternions	invalid (must be unit quaternions)
	timestamps	ok
stats:
	v_avg (km/h)	1.595637
	v_avg (m/s)	0.443233
	v_max (km/h)	7.998436
	v_max (m/s)	2.221788
	v_min (km/h)	0.000000
	v_min (m/s)	0.000000

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: v1.7.1
* Python version shown by `evo pkg --pyversion`: 2.7.15+
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04.1 LTS
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```

{
    "console_logging_format": "%(message)s", 
    "euler_angle_sequence": "sxyz", 
    "global_logfile_enabled": false, 
    "plot_axis_marker_scale": 0.0, 
    "plot_backend": "Qt5Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        7, 
        6
    ], 
    "plot_fontfamily": "serif", 
    "plot_fontscale": 1.2, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1, 
    "plot_multi_cmap": "none", 
    "plot_reference_alpha": 0.5, 
    "plot_reference_color": "black", 
    "plot_reference_linestyle": "--", 
    "plot_seaborn_palette": "deep6", 
    "plot_seaborn_style": "whitegrid", 
    "plot_split": false, 
    "plot_statistics": [
        "rmse", 
        "median", 
        "mean", 
        "std", 
        "min", 
        "max"
    ], 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_alpha": 0.75, 
    "plot_trajectory_cmap": "jet", 
    "plot_trajectory_linestyle": "-", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "ros_map_unknown_cell_value": 205, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true
}

```
Thanks in advance for your time.

MichaelGrupp, 2020-02-28:
It uses numpy's allclose to check if the quaternions are unit quaternions.

The default tolerance is quite low (1e-8), so it 's not necessarily something to worry about.

https://docs.scipy.org/doc/numpy/reference/generated/numpy.allclose.html
https://github.com/MichaelGrupp/evo/blob/4a80b43aa20c7b499215610567ea395e89eaa500/evo/core/trajectory.py#L202

closed 2020-03-11 by MichaelGrupp

IoannisKaragiannis, 2020-03-11:
Thanks for your reply ;)