MichaelGrupp/evo #249: Theory about error metrics

ricber, 2020-02-19:
Hi, I would like to better understand the concepts behind the error metrics. Especially, I would like to know how you condensed the x, y, z components in one single number for both the translation and rotation error metrics. Actually, for the translation part is clear, you are just using the norm of the translation difference between the estimated and reference poses. 

What about the rotation error? I see that you are condensing the 3x3 rotation matrix in a single number with the log operator in SO(3) group. Then you apply an angle() function. Could you explain better what is the meaning of applying the log_SO(3) function and what the angle() function does? Also, I would like to know the concept behind the unitless rotation error metrics. I checked the two following papers but I didn't find an answer. Could you suggest some papers where these concepts are explained? Thank you very much :)

> Rainer Kümmerle, Bastian Steder, Christian  Dornhege, Michael Ruhnke, Giorgio Grisetti, Cyrill Stachniss, and  Alexander Kleiner. On measuring the accuracy of SLAM algorithms. Autonomous Robots, 27(4):387–407, 2009.

> Jürgen Sturm, Nikolas Engelhard, Felix Endres, Wolfram Burgard, and  Daniel Cremers. A benchmark for the evaluation of RGB-D SLAM systems. In 2012 IEEE/RSJ International Conference on Intelligent Robots and Systems, pages 573–580. IEEE, 2012.

MichaelGrupp, 2020-02-28:
This paper describes different rotation metrics very well: http://www.cs.cmu.edu/~cga/dynopt/readings/Rmetric.pdf
(some prior knowledge of Lie algebra and different rotation representations is helpful)

> Could you explain better what is the meaning of applying the log_SO(3) function and what the angle() function does?

It corresponds to extracting the angle component of the logarithmic map of the rotation matrix. See also eq. 23 of the paper.

> Also, I would like to know the concept behind the unit-less rotation error metrics

They measure the deviation from the identity matrix with the Frobenius norm, which is a valid metric. Equation 23 of the linked paper does the same. This is interesting in theory, but usually not so useful in real life.

All of these rotation metrics are similar. If you compare the values of the angle metrics and the unit-less one, they only differ by a constant factor.

ricber, 2020-02-28:
Thank you very much Michael! Your answers are very useful. Just a last question. Does the angle component of the logarithmic map correspond to the angle component of the axis-angle representation of the rotation matrix?

MichaelGrupp, 2020-03-11:
Yes https://en.wikipedia.org/wiki/Axis%E2%80%93angle_representation#Relationship_to_other_representations

closed 2020-03-23 by MichaelGrupp