MichaelGrupp/evo #255: Wrong map setting in evo_traj bag --ros_map_yaml

Krada-Jhin, 2020-03-08:
**Description:**
I gonna to load ros map in plot , and I find when I load this map ,its initial pose it its origin in yaml.
However , this intial pose of map dose not match the trajectory in odom or other topic.
 

**Command:**
```
evo_traj bag badmap.bag --all_topics -p --plot_mode xy --ros_map_yaml test_room1_origin.yaml 
```

**Console output:**
```

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version       1.7.1: 
* Python version 2.7.12: 
* Ubuntu 16.04
* did you change the source code? (
[image]
no): 
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```

Krada-Jhin, 2020-03-08:
[image]

MichaelGrupp, 2020-03-11:
The "origin" in the YAML file describes the pose of the lower left corner with respect to the map TF frame (http://wiki.ros.org/map_server#YAML_format). This should be saved correctly with tools like the map_saver in ROS. Otherwise, check it if you create the YAML / PGM files differently.

If your trajectory poses are relative to this map TF frame, it will be displayed correctly here.

closed 2020-03-24 by MichaelGrupp