MichaelGrupp/evo #256: Can I change the 'Error mapped onto trajectory' when I use evo_ape

bryantaoli, 2020-03-15:
Can I change the 'Error mapped onto trajectory' when I use evo_ape
[image]

MichaelGrupp, 2020-03-16:
Can you be a little more specific what you want to change?

bryantaoli, 2020-03-16:
[image]
I want to change this.

MichaelGrupp, 2020-03-16:
Then you need to edit the code here: https://github.com/MichaelGrupp/evo/blob/master/evo/common_ape_rpe.py#L143

Installing the code from source is described in the README

bryantaoli, 2020-03-16:
But I install the evo from pip，does it mean I should uninstall the evo and install it again by source ？

MichaelGrupp, 2020-03-16:
Yes.

closed 2020-03-24 by MichaelGrupp