MichaelGrupp/evo #257: how can i use evo to draw a trajectory in three different color

1612190130, 2020-03-16:
I was testing image segmentation on Kitti. So i want to visualize the trajectory in different color for every cluster on one window.I can not  find any information of  how to invoke the interface of evo in code to reach my objective.

MichaelGrupp, 2020-03-18:
If you want to plot only different segments of the trajectory in different colors, you could just split them up across different trajectory files (one for each segment). If you load and plot this, it will be shown in different colors for each segment.

If you need more customization you can take a look at the internal function that can plot a trajectory in multiple colors based on a set of values and a colormap:
https://github.com/MichaelGrupp/evo/blob/master/evo/tools/plot.py#L365

1612190130, 2020-03-23:
> If you want to plot only different segments of the trajectory in different colors, you could just split them up across different trajectory files (one for each segment). If you load and plot this, [...]

Thanks!

closed 2020-03-23 by MichaelGrupp