MichaelGrupp/evo #258: Some questions about the evaluation indicators

zcg3648806, 2020-03-20:
When I run evo_ape, I can get:

Aligning using Umeyama's method... (with scale correction)
Rotation of alignment:
[[ 0.99966481  0.01007943  0.02384699]
 [-0.0095949   0.99974696 -0.02034599]
 [-0.02404603  0.02011036  0.99950856]]
Translation of alignment:
[-0.48982277  0.97985428  4.66481522]
Scale correction: 16.0559813338
--------------------------------------------------------------------------------
Compared 1511 absolute pose pairs.
Calculating APE for translation part pose relation...
--------------------------------------------------------------------------------
APE w.r.t. translation part (m)
(with Sim(3) Umeyama alignment)

       max	11.813678
      mean	5.250972
    median	4.376774
       min	0.109121
      rmse	6.126788
       sse	56719.204065
       std	3.156710

I want to know what do the Rotation of alignment and the Translation of alignment mean?
Besides,there is only ape  translation part ,how do I know the ape of rotation?

MichaelGrupp, 2020-03-20:
> what do the Rotation of alignment and the Translation of alignment mean?

They represent the 3D transformation that aligns the trajectory to the reference trajectory.

> how do I know the ape of rotation?

There is a `--pose_relation` option. For rotation angle you can use `--pose_relation angle_rad` or `--pose_relation angle_deg` for degrees.

zcg3648806, 2020-03-21:
@MichaelGrupp Thank you very much! I get it.
i want to ask you another question.

[image]

This is one of the outcomes of evo_res and what does it mean?Thank you!

MichaelGrupp, 2020-03-21:
It is a box plot of the error values of all poses: https://en.wikipedia.org/wiki/Box_plot

zcg3648806, 2020-03-22:
Thanks for your reply! Please forgive me for having a new problem.
I read some of the papers on slam, and I found that in these papers the performance of slam was evaluated with  the average relative translation （%） and rotation （deg/m）,  It's just like the kitti list shows：
[image]
How do evo get this result？Thank you！

zcg3648806, 2020-03-25:
@MichaelGrupp I'm sorry to bother you again. According to my understanding, both ape and rpe are comparing the errors between frames, and their units are m or deg. However, how did the average pose errors per meter or per 100 meters mentioned in the paper and kitti's official website be obtained?

MichaelGrupp, 2020-03-26:
I'm pretty sure there is code & documentation for that from the KITTI dataset. You could configure evo to do something similar, but keep on mind it is not the official KITTI tool.

closed 2020-03-26 by MichaelGrupp