MichaelGrupp/evo #260: RPE result has something wrong on TUM dataset

shelumielzhang, 2020-03-26:
**Description:**
I just want to evaluate my result on TUM dataset, but the rpe result looks something wrong

**Command:**
```
evo_rpe tum groundtruth.txt results.txt -p -a
```

**Console output:**
```

PE w.r.t. translation part (m)
for delta = 1 (frames) using consecutive pairs
(with SE(3) Umeyama alignment)

       max	0.419719
      mean	0.007207
    median	0.005317
       min	0.000560
      rmse	0.013662
       sse	0.308163
       std	0.011606
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```

MichaelGrupp, 2020-03-26:
Please check start and end timestamp of your input data. Are they sorted?

shelumielzhang, 2020-03-26:
> Please check start and end timestamp of your input data. Are they sorted?

Yes, I have checked the timestamp, and here is my input file
[results.txt](https://github.com/MichaelGrupp/evo/files/4386062/results.txt)

MichaelGrupp, 2020-03-26:
No, you have 2 trajectories in the same file.

Look at this check and the plots
```
evo_traj tum results.txt --full_check --plot
```

MichaelGrupp, 2020-03-26:
It should be fine if you fix your file.

closed 2020-03-26 by MichaelGrupp

shelumielzhang, 2020-03-26:
> No, you have 2 trajectories in the same file.
> 
> Look at this check and the plots
> 
> ```
> evo_traj tum results.txt --full_check --plot
> ```

Oh, thank you so much! I just use associate.py to align rgb and depth, but why there are two same  sequenses in associated.txt?

MichaelGrupp, 2020-03-26:
I don't know and I can't help you with this, you need to debug this yourself.