MichaelGrupp/evo #265: Evo ape rotation part stops midway

alexsimon1, 2020-04-19:
hello,

during my evaluation of the absolute error (rotation part) my estimatet trajectory stopps midway (picture1). The evaluation of the traslation part works just fine (picture2).

[image]

[image]

What could be the reason?

Thank you.

MichaelGrupp, 2020-04-20:
> What could be the reason?

Please provide more information & data to reproduce this.

alexsimon1, 2020-04-20:
Hello, 

i can provide the estimate and reference tum files.
[est.txt](https://github.com/MichaelGrupp/evo/files/4502187/est.txt)
[ref.txt](https://github.com/MichaelGrupp/evo/files/4502188/ref.txt)

MichaelGrupp, 2020-04-20:
The `est.txt` that you uploaded just does not contain more poses than that.

See `evo_traj tum est.txt --plot_mode xy -p`

alexsimon1, 2020-04-20:
Oh right,

there was a recording error for the estimate.

Thanks you.

closed 2020-04-20 by alexsimon1