MichaelGrupp/evo #266: Disable legend

alexsimon1, 2020-04-20:
Hello,

I can remove spefic trajectorys from the plot, is it possible to remove the corresponding parts of the legend?

MichaelGrupp, 2020-04-22:
> I can remove spefic trajectorys from the plot

How do you do that?

alexsimon1, 2020-04-22:
Hello, 

I mean visually removal, by changing the line style to none.

MichaelGrupp, 2020-04-22:
So, if you want to remove the trajectories completely from the plot - why don't you just not load them in the first place?

alexsimon1, 2020-04-22:
I wrote an automatic skript that collects all estimated trajectories and plots them. It would be a nice option to just blend out the legends/lines to quickly compare only a couple of trajectorys. I must say i have 48 trajectorys

MichaelGrupp, 2020-04-22:
If you don't pass the "label" argument there should be no legend: https://github.com/MichaelGrupp/evo/blob/7382aa130a9b86891d621ea8dd8204a55678810b/evo/tools/plot.py#L316

closed 2020-04-22 by alexsimon1