MichaelGrupp/evo #267: Draw Line between compared Poses

alexsimon1, 2020-04-28:
Hello,

is it possible to draw lines between the compared poses?
The viewer could then easily see where the estimated Pose should be.

Thank You.

MichaelGrupp, 2020-04-30:
Should be easy to add, in the end it's just adding a line collection with line segments going from one pose to the other trajectory's pose.

I will consider adding this as a new feature in this package. Until then, adding it as a custom implementation in your own code shouldn't be too hard either using https://matplotlib.org/3.1.1/api/collections_api.html#matplotlib.collections.LineCollection

MichaelGrupp, 2020-05-08:
Keeping this open as a feature request.

MichaelGrupp, 2020-05-17:
Added in the latest version:

https://github.com/MichaelGrupp/evo/wiki/Plotting#pose-correspondence-markers

closed 2020-05-17 by MichaelGrupp