MichaelGrupp/evo #269: Start plot times from zero

mcamurri, 2020-05-12:
Thank you for this tool, it is really great, very well made and documented.

When using `evo_traj` with the data in TUM format, the times start from the Epoch so they contain huge numbers, like `1582234689.786309004`. 

This is very inconvenient because the time plots are not clear:
[image]

I know there is a `--t_offset` option, but this is to apply an offset to the trajectories to be printed, without affecting the reference trajectory. I believe it is used when you know there is a delay between a trajectory and its reference. Also, the offset would be manual.

What would be nice to have is an option like `--plot-time-from-zero` that internally uses the original times  for calculation (e.g. for sync) but when plotting have the time that starts from zero automatically.

It shouldn't be hard to implement, I guess.

MichaelGrupp, 2020-05-14:
Subtracting the value of the first timestamp of the reference from all synced trajectories should be enough. There was actually a feature for that, but I removed it because it looked weird under some circumstances. Reviving it as an option would be ok I guess :wink:

mcamurri, 2020-05-14:
That would be great thanks!

MichaelGrupp, 2020-05-17:
This is now possible with the latest version and `--plot_relative_time` :tada:

closed 2020-05-17 by MichaelGrupp