MichaelGrupp/evo #273: How to cancel the influence of a certain axis data

Octolsineb, 2020-05-17:
I want to know the x,y axis error between my data set and the true value,and ignore z axis.
Just like the following picture.
[image]

Octolsineb, 2020-05-17:
The maximum value in the figure below is obviously on the z axis, but I just want to know the deviation in the xy direction and plot it
[image]

MichaelGrupp, 2020-05-17:
You need to project your input data to the 2D XY plane, so the z-component is zero. Then it won't have an influence on the metrics. 

Besides that there's no easy way to _generally_ ignore Z in this software. It's designed for 3D data and changing it would be a lot of work.

MichaelGrupp, 2020-05-18:
(both reference and estimated trajectory would need to be projected to 2D in this case, of course)

Octolsineb, 2020-05-18:
> (both reference and estimated trajectory would need to be projected to 2D in this case, of course)

thx!!! Maybe I know how to deal with it！Really appreciate it! :)

closed 2020-05-18 by Octolsineb