MichaelGrupp/evo #274: export evo_traj as .eps instead of .png

cnqiangfu, 2020-05-21:
Hi, i am writing a paper using latex, i find png is not clear enough. There are two options to cope with it. But i donot know how to do it by using the wonderful tool:
(1) save a high DPI imgae.png, like plt.savefig("image", dpi = 600) 
(2)export the object as image.eps.
Thanks for your help.

cnqiangfu, 2020-05-22:
I have solve it with a small change in /tools/plot.py.

MichaelGrupp, 2020-05-22:
If you use the command line tools with `--save_plot` or use the `tools.plot.PlotCollection` class in your own code, then you can change the format setting:

```
evo_config set plot_export_format eps
```

---

If you have simple matplotlib figures you want to save, you can use [`savefig`](https://matplotlib.org/3.1.1/api/_as_gen/matplotlib.pyplot.savefig.html) as usual.

cnqiangfu, 2020-05-23:
> If you use the command line tools with `--save_plot` or use the `tools.plot.PlotCollection` class in your own code, then you can change the format setting:
> 
> ```
> evo_config set plot_export_format eps
> ```
> 
> It's used here:
> 
> https://github.com/MichaelGrupp/evo/blob/00450dfb37e4ec36246ba25da59a3807c296083e/evo/tools/plot.py#L207
> 
> If you have simple matplotlib figures you want to save, you can use [`savefig`](https://matplotlib.org/3.1.1/api/_as_gen/matplotlib.pyplot.savefig.html) as usual.

closed 2020-05-23 by cnqiangfu