MichaelGrupp/evo #275: Change the ros topic name when doing evo

CrazyPopLin, 2020-05-27:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**  

First, I just want double check if there is a similar feature already, maybe I did't find it. 

The solution:   
when I have some odometries to plot, but most of them have ugly names like **/test/odom_correct** or **/nav/processed/odom**, which not good for paper or report. Although, I can use ROS bag API to rename the topic to **odometry_IMU** and **odometry_reference**. If we have rename option, that's well be perfect. Thanks.

MichaelGrupp, 2020-05-27:
There is a very nice tool for editing ROS bags: https://github.com/facontidavide/rosbag_editor 
It allows also renaming topics very easily. I installed it on ROS melodic with
```
sudo apt install ros-melodic-rosbag-editor
```

Alternatively, you can save the trajectories to individual files in TUM format with `--save_as_tum`, rename the files to a new name and then load into `evo_* tum ...` instead of `bag`.

I would prefer to not add any renaming logic to evo, because it would get a bit complex for multiple 
trajectories and there are better tools for this task (like the rosbag_editor).

CrazyPopLin, 2020-05-27:
thx, that's helped !

closed 2020-05-27 by CrazyPopLin