MichaelGrupp/evo #281: Results logging - Global trajectory metrics such as path length, speed

YoshuaNava, 2020-06-09:
Hi,
Thank you for this great package. It's really useful for measuring localization performance.

I was wondering if global trajectory metrics such as path length, average speed, and others are currently saved by evo_ape/rpe/res.

* If so, in which of the results file?
* If not, would it be possible to add these metrics to the result logfiles and the plots?

Thank you in advance.

Yoshua Nava

MichaelGrupp, 2020-06-12:
For reference: the data you mention is available for each trajectory as dictionaries via the `get_{infos, statistics}` methods: https://github.com/MichaelGrupp/evo/blob/master/evo/core/trajectory.py#L283

---

> I was wondering if global trajectory metrics such as path length, average speed, and others are currently saved by evo_ape/rpe/res.

No. The result files generated by these tools contain only the statistics and raw values of the error metric that was calculated, and optionally the trajectories: https://github.com/MichaelGrupp/evo/wiki/Formats#result-zip-files-for-evo_res

That being said, from a technical POV it is however possible to add more data to the statistics of a "`Result`" (in fact, arbitrary [dicts](https://github.com/MichaelGrupp/evo/blob/master/evo/core/result.py#L83) / arrays) that would get also [serialized/deserialized](https://github.com/MichaelGrupp/evo/blob/master/evo/tools/file_interface.py#L327).

---

> If not, would it be possible to add these metrics to the result logfiles and the plots?

I think they don't belong in the result files, at least no the ones generated by `evo_{ape, rpe}`, because this data doesn't come from the pose error calculation.

Since the data directly comes from the trajectories, it would rather be a job for `evo_traj`. A somewhat related issue is this: #238

MichaelGrupp, 2020-06-28:
`--save_table` for `evo_traj` was added in the latest version. It saves the data you mentioned.

closed 2020-06-28 by MichaelGrupp

YoshuaNava, 2020-06-28:
Sorry for the delay in responding. 

> --save_table for evo_traj was added in the latest version. It saves the data you mentioned.
 
Thank you very much!